360Controller/360Controller #701: Amazon Basic Xbox One controller amazon.com/dp/B01M4NISMM

cjac, 2018-06-13:
### Type of Controller
Xbox One
amazon.com/dp/B01M4NISMM

### OS Version
10.13.5

### Driver Version
0.16.8

### Connection Method
wired

### Device Name and Info

Xbox ONE Pro Ex controller:

  Product ID:	0x543a
  Vendor ID:	0x24c6
  Version:	1.01
  Serial Number:	0000E74A2778F41B
  Speed:	Up to 12 Mb/sec
  Manufacturer:	BDA
  Location ID:	0x14410000 / 7
  Current Available (mA):	500
  Extra Operating Current (mA):	0

    | +-o Xbox ONE Pro Ex controller@14410000  <class AppleUSBDevice, id 0x10000045d, registered, matched, active, busy 0 (45 ms), retain 12>
    |     {
    |       "sessionID" = 2446235178
    |       "iManufacturer" = 1
    |       "bNumConfigurations" = 1
    |       "idProduct" = 21562
    |       "bcdDevice" = 257
    |       "Bus Power Available" = 250
    |       "USB Address" = 7
    |       "bMaxPacketSize0" = 64
    |       "iProduct" = 2
    |       "iSerialNumber" = 3
    |       "bDeviceClass" = 255
    |       "Built-In" = No
    |       "locationID" = 339804160
    |       "bDeviceSubClass" = 255
    |       "bcdUSB" = 512
    |       "USB Product Name" = "Xbox ONE Pro Ex controller"
    |       "PortNum" = 1
    |       "non-removable" = "no"
    |       "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
    |       "bDeviceProtocol" = 255
    |       "IOUserClientClass" = "IOUSBDeviceUserClientV2"
    |       "IOPowerManagement" = {"DevicePowerState"=0,"CurrentPowerState"=3,"CapabilityFlags"=65536,"MaxPowerState"=4,"DriverPowerState"=3}
    |       "Device Speed" = 1
    |       "USB Vendor Name" = "BDA"
    |       "idVendor" = 9414
    |       "IOGeneralInterest" = "IOCommand is not serializable"
    |       "USB Serial Number" = "0000E74A2778F41B"
    |       "IOClassNameOverride" = "IOUSBDevice"
    |     }

### Response of Controller
No light.  No vibration.  The only way I know it's connected is because it shows up in System Information and ioreg -p IOUSB -l -w 0

### Response in Preference Pane
controller is not detected: No devices found

### Detailed Explaination of Issue
Device not detected in preference pane.  Perhaps the vendor ID and product ID need to be added to the source?

cjac, 2018-06-13:
looks like this product/vendor code is already included:

`                <key>PowerAMiniXboxOne2</key>
                <dict>
                        <key>CFBundleIdentifier</key>
                        <string>com.mice.driver.Xbox360Controller</string>
                        <key>IOCFPlugInTypes</key>
                        <dict>
                                <key>F4545CE5-BF5B-11D6-A4BB-0003933E3E3E</key>
                                <string>360Controller.kext/Contents/PlugIns/Feed
back360.plugin</string>
                        </dict>
                        <key>IOClass</key>
                        <string>Xbox360Peripheral</string>
                        <key>IOKitDebug</key>
                        <integer>65535</integer>
                        <key>IOProviderClass</key>
                        <string>IOUSBDevice</string>
                        <key>idProduct</key>
                        <integer>21562</integer>
                        <key>idVendor</key>
                        <integer>9414</integer>
                </dict>
`

cjac, 2018-06-13:
I've attached my legitimate, original, manufactured by MS Xbox 360 controller, and the guide light blinks, but it also is not found, ie "Device not found" is the only option in the drop-down menu.

FranticRain, 2018-06-16:
I need you to open up your terminal and run `sudo kextload /Library/Extensions/360Controller.kext` and tell me what it says.

If it says `/Library/Extensions/360Controller.kext failed to load - (libkern/kext) system policy prevents loading; check the system/kernel logs for errors or try kextutil(8).` Then you'll need to allow the driver in the `Security & Privacy` section of your `System Preferences`.
[image]

Otherwise, I'll have to give the error a look to see what the problem is.

FranticRain, 2018-12-04:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-12-04 by FranticRain

zdinardo, 2020-01-15:
Hi! I have the same controller issue as OP. Different serial number/location info, but everything else is pretty much the same.  I'm on MacOS 10.15.2 and am on driver 1.0.0-alpha_5 (as well as 0.9.1). When I run  `sudo kextload` with the filename, nothing shows up. Happy to create a new issue if that's where this belongs.

FranticRain, 2020-01-15:
@zdinardo Please make a new issue.