360Controller/360Controller #703: Xbox One controller not showing up in 360controller menu

TD1337, 2018-06-19:
### Type of Controller
Xbox One
### OS Version
V10.13.3
### Driver Version
Dont know what driver is or version
### Connection Method
Wired USB cord.
### Device Name and Info
Cant find this on the readme, pls explain
<br/>Product ID: 0x02dd
<br/>Vendor ID:0x045e  (Microsoft Corporation)

### Response of Controller
Vribrates when plugged in but the light doesnt stop flashing, turns off after 20ish seconds
### Response in Preference Pane
The controller shows up in the system information but not the system preferences xbox360 thing. The controller doesnt affect anything on computer with joysticks and such.

### Detailed Explaination of Issue
When connected to the computer the controller vibrates and flashes but does nothing else, ive checked that i dont have a mouse filter/changer and I have allowed drew mills, if you need other information just ask bc i dont know much about controllers and drivers.
Thanks
Tom

FranticRain, 2018-06-19:
The version number of the driver can be found in the "About" tab of the preference pane. Also, you said that you can't find some part of the device name and info, yet you have values in there? The README says how to obtain this information: https://github.com/360Controller/360Controller#how-do-i-find-my-vendor-id-and-product-id

I need this additional information, but in the mean time, attempt the debugging as follows:

I need you to open up your terminal and run `sudo kextload /Library/Extensions/360Controller.kext` and tell me what it says.

If it says `/Library/Extensions/360Controller.kext failed to load - (libkern/kext) system policy prevents loading; check the system/kernel logs for errors or try kextutil(8).` Then you'll need to allow the driver in the `Security & Privacy` section of your `System Preferences`.
[image]

Otherwise, I'll have to give the error a look to see what the problem is.

FranticRain, 2018-12-04:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-12-04 by FranticRain