360Controller/360Controller #704: XBOX 360 knock-off controller not working on Mac

metuome95, 2018-06-19:
xbox 360 knock off:

https://www.amazon.com.au/Kycola-Controller-controller-Gamepad-PC（Deep/dp/B01IN2P12S/ref=sr_1_14?ie=UTF8&qid=1529409678&sr=8-14&keywords=xbox+game+controller

current os:
Mac OS high Sierra 10.13.2

used:
Xbox 360 Controller Driver
Version 0.16.8. downloaded from GIT hub

connection:
wired - usb

Inno GamePad..:
 Product ID:	0xfa01
  Vendor ID:	0x1bad
  Version:	1.10
  Serial Number:	0000001
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Inno GamePad..
  Location ID:	0x14100000 / 4
  Current Available (mA):	500
  Extra Operating Current (mA):	0

flashing centre lights -- on a PC connects and works immediately-- works in big pic mode in steam. the centre light stops blinking just is solid

however in the Mac,  it is not recognised at all....not recognised by the GIT drivers, or even joystick mapper, when about Mac>system report> usb, 
is read as: Inno Gamepad

trying to find a driver to make it work, thanks all

FranticRain, 2018-06-19:
I need you to open up your terminal and run `sudo kextload /Library/Extensions/360Controller.kext` and tell me what it says.

If it says `/Library/Extensions/360Controller.kext failed to load - (libkern/kext) system policy prevents loading; check the system/kernel logs for errors or try kextutil(8).` Then you'll need to allow the driver in the `Security & Privacy` section of your `System Preferences`.
[image]

Otherwise, this issue will be closed down, because we do not support counterfeit controllers.

FranticRain, 2018-08-20:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-08-20 by FranticRain