360Controller/360Controller #705: Macbook not recognising wired Xbox controller

studsr, 2018-06-23:
### Type of Controller
Original Xbox

### OS Version
High Sierra 10.13.5

### Driver Version
0.16.8

### Connection Method
Wired

### Detailed Explaination of Issue
I plug controller in, the green light flashes for a second, so it looks like the Macbook can se et, however your software says @no controller found"

Any help much appreciated, I see others are having similar problems,

Andy.

studsr, 2018-06-25:
Here's the info you requested Drew.

[image]

Thanks,

Andy

FranticRain, 2018-06-28:
I need you to open up your terminal and run `sudo kextload /Library/Extensions/360Controller.kext` and tell me what it says.

If it says `/Library/Extensions/360Controller.kext failed to load - (libkern/kext) system policy prevents loading; check the system/kernel logs for errors or try kextutil(8).` Then you'll need to allow the driver in the `Security & Privacy` section of your `System Preferences`.
[image]

Otherwise, this issue will be closed down, because we do not support counterfeit controllers.

FranticRain, 2018-10-21:
Counterfeit controllers are devices that use the product and vendor IDs of the official Xbox 360 controller in order to try and spoof hardware into recognizing them as official controllers. They are either cheap imitations of the original controller, or unlicensed Chinese hacks. Either way, they violate the USB spec, and therefore cannot be individually supported by the driver.

This issue is being closed due to inactivity.

closed 2018-10-21 by FranticRain