360Controller/360Controller #706: Driver not working on Fortnite with third-party controller

nluke94, 2018-06-24:
Xbox One Power A Wired Controller

MacOS Sierra 10.12.6

360Controller 0.16.8

Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: PowerA Xbox One Pro Ex Controller
<br/>Product ID: 0x543a
<br/>Vendor ID:0x24c6

### Response of Controller
No lights show up but some controls work within the game

### Response in Preference Pane
The Xbox 360 Controllers option shows up in system preferences but it is grayed out and I can't select it. 

### Detailed Explaination of Issue
I've only tried to use it within Fortnite on my Mac. Most buttons provide a response but the responses don't match up with the button being pressed. The right stick does not work at all, but the right trigger causes the view to be changed as if I have the right stick pushed down constantly, even if I let go of the right trigger. Please let me know if you need any more info, I'm a bit new to this. Thanks a lot

FranticRain, 2018-06-24:
Please ensure that the driver has been allowed in System Preferences -> Security and Privacy -> General. This approval UI is only present in the Security & Privacy preferences pane for 30 minutes after the alert. Until you approve the kext, future load attempts will cause the approval UI to reappear but will not trigger another user alert. This should resolve the issues with the preference pane being grayed out.

[image]

Your issues with Fortnight are probably game specific. The driver does not provide any sort of compatibility for individual games beyond what is provided in the preference pane. The driver simply sends button data to the operating system. Once it does that, control is out of the driver's hands. Since Apple does not provide any standardized means of interacting with controllers in macOS, every game uses a different method to read controller data. We cannot provide individual fixes for each game within the controller. That simply isn't feasible. It is the responsibility of game developers to support controllers, not the other way around. We cannot and will not provide support for individual games. If you are using an Xbox One controller, we can recommend that you use the "Pretend to be an Xbox 360 Controller" option in the "Advanced" menu for better game compatibility. Beyond that, there is nothing we can do.

nluke94, 2018-06-24:
Thanks a lot for the response. I saw your approval UI suggestion in some of your previous responses. However, I am not seeing that "allow" option. I have it set to "Anywhere" which I'm guessing is why I never got the user alert. 

[image]

As far as the game specific issue goes, that's fine. I just want to get the preference pane to not be grayed out, so that I can try the "Pretend" option you mentioned (which I assume is in system preferences) . Is there anything else you think I could do to achieve this? Thanks again

FranticRain, 2018-06-24:
That approval UI doesn't exist in 10.12. I should have noticed that in your original post, sorry. I don't know why that happens. It's probably due to some sort of permissions settings on your computer, but I wouldn't know for sure. It's never been fixed by anyone and I've never experienced it, so I don't know of any method to fix it.

JohnCalvinHobbes, 2018-07-18:
Made an account specifically to comment here. I had the exact same issue as you with the same brand of controller: got the controller specifically to play Fortnite because I can't get the hang of the M&K setup, but then everything was out of whack when I tried to use it. Now, my setup is slightly different than yours, which means your solution will probably be different, but my hope is that it might point you or someone else more knowledgeable than me in the right direction. I play on my Macbook through a Parsec cloud gaming computer that I rent (which is very cheap and very awesome), and every time I connect to play (a bit of an inconvenience), I 
(1) plug in my controller, 
(2) open Fortnite to the initial screen showing options for PvE and Battle Royale, 
(3) remap my controller using Parsec's built in controller-mapping software, 
(4) unplug and re-plug the controller. 
After that, my controller works fine and I only have my own inability to blame when I die. Step (3) is where we our paths diverge, but maybe if you can find a remapping tool that recognizes the controller, this will help. 

Also, while I don't have the same System Preferences issue that you do, maybe try uninstalling and then reinstalling in safe mode with the controller connected and your firewall and antivirus disabled? This approach, for some reason, has worked for me in other contexts in the past. Hope you get connected!

kurtfm, 2018-09-18:
Thanks @JohnCalvinHobbes this worked for me.

FranticRain, 2019-01-15:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-01-15 by FranticRain