360Controller/360Controller #708: xbox one controller shows is preference but not in driver

lukenic, 2018-06-26:
### Type of Controller
 Xbox One 

### OS Version
OS high Sierra 10.13.5

### Driver Version
360Controller 0.16.8

### Connection Method
 Wired

### Device Name and Info
>Device Name:
<br/>Product ID: 0x02d1
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller
no response from the controller 

### Response in Preference Pane
nothing 

### Detailed Explaination of Issue
I connected my xbox one controller to my Mac and it shows in the preference system of the Mac but the controller doesnt show in the 360controller driver 

can you help me out ? let me know if you miss any information

FranticRain, 2018-06-28:
I need you to open up your terminal and run `sudo kextload /Library/Extensions/360Controller.kext` and tell me what it says.

If it says `/Library/Extensions/360Controller.kext failed to load - (libkern/kext) system policy prevents loading; check the system/kernel logs for errors or try kextutil(8).` Then you'll need to allow the driver in the `Security & Privacy` section of your `System Preferences`.
[image]

Otherwise, this issue will be closed down, because we do not support counterfeit controllers.

lukenic, 2018-06-28:
Ok it is working, I allowed the system software and this is working 

Thank you

closed 2018-06-29 by FranticRain