360Controller/360Controller #709: Request: Allow right stick to be custom mapped

daxfox, 2018-06-29:
### Type of Controller
Xbox 360

### OS Version
10.12.6

### Driver Version
0.16.8

### Connection Method
Wireless Adapter

### Device Name and Info
N/A

### Response of Controller
Controller responds appropriately as expected.

### Response in Preference Pane
All the controls react as expected, and the animations all work in the Controller Test panel.

### Detailed Explaination of Issue
I'm trying to get the right stick motions to be recognized by MAME OS X (.135).
By default, MAME .135 is not seeing the right joystick input.
However, it has mapped that input to the 4 buttons A B X Y for DOWN RIGHT LEFT UP, and those work correctly.

My request is for the user to be able to configure the right stick so that it outputs custom key bindings.  If the user presses the Right Joystick to the Right, send a custom keypress.  Same thing for the other three directions.  Then MAME can be configured to recognize these right stick motions as keypresses and my problem is solved.

The reason for this feature request is that I'm not yet sure .135 is configured to work with the RXAXIS and RYAXIS joystick input (e.g.: JOYCODE_1_RXAXIS_POS_SWITCH, JOYCODE_1_RYAXIS_NEG_SWITCH).  If anyone knows how to fix this by configuration alone in .135 let me know, thank you!

k1down, 2018-06-29:
working as intended

FranticRain, 2018-06-29:
This is outside the scope of this driver. Rebinding has long be regarded as a mistake in the driver, but we are remiss to remove it, as it might result in some users using older versions of the driver when they should upgrade.

No current member of the team would like to add this feature, and instead, we recommend third party tools that were built explicitly for this purpose, such as [Enjoyable](https://yukkurigames.com/enjoyable/) and [Joystick Mapper](http://joystickmapper.com/).

However, if you or someone else would like to build and support this feature, you are more than welcome to.

MrLavender, 2018-11-12:
This issue also affects some older games, most notably the Grand Theft Auto 3 series (original, Vice City, and San Andreas).

While it is kind of fixable outside the driver that is a pain to be honest, and there is a simple solution which is to allow swapping the right stick <-> trigger axes. The driver already allows swapping X <-> Y axes, that functionality is useless to me so I'm running a custom kext with the method changed to swap right stick and trigger instead. I've been making this change for years now and never had a problem with it (apart from having to run with SIP disabled because I don't have a driver signing certificate).

    void Xbox360ControllerClass::remapAxes(void *buffer)
    {
        XBOX360_IN_REPORT *report360 = (XBOX360_IN_REPORT*)buffer;
    /*
        XBOX360_HAT temp = report360->left;
        report360->left = report360->right;
        report360->right = temp;
    */
        XBOX360_HAT temp = report360->right;

        // Stick axes are signed 16-bit, trigger axes are unsigned 8-bit

        // 0, 128, 255 -> -32768, 0, +32512
        report360->right.x = (report360->trigL - 128) * 256;
        report360->right.y = (report360->trigR - 128) * 256;

        // -32768, 0, +32767 -> 0, 128, 255
        report360->trigL = (temp.x / 256) + 128;
        report360->trigR = (temp.y / 256) + 128;
    }

FranticRain, 2018-11-13:
If it works so well, then you're more than welcome to create a merge request with that, wrapped in a property, with a setting in the preference pane. And you'll be on the hook to support it.

daxfox, 2018-11-13:
The reason I originally drafted this improvement request is so that I can simultaneously use the left stick and the right stick controls at the same time in emulated MAME games like Smash TV, Robotron 2048, and other two-joystick games.

Presently, the MAME app I’m using with the joystick driver doesn’t recognize the right joystick and left joystick at the same time for  these two-joystick games, but I know that if the right stick can be remapped to bind to the proper movement keys in the driver, then those key presses can be finally be sent by the driver and recognized by MAME.

So, to sum up, It’s not a matter of swapping left and right sticks, but making both visible at the same time through allowing key bindings on the right stick, which the driver doesn’t currently support.

MrLavender, 2018-11-13:
> If it works so well, then you're more than welcome to create a merge request with that, wrapped in a property, with a setting in the preference pane. And you'll be on the hook to support it.

Ok I may do that when I get some time :)

MrLavender, 2018-11-14:
> The reason I originally drafted this improvement request is so that I can simultaneously use the left stick and the right stick controls at the same time in emulated MAME games like Smash TV, [...]

Yes I see, I've just tried MAMEOSX 0.135 (I assume [this is the one you're using](https://sourceforge.net/projects/mameosx/)?) and it's not the same problem. It just doesn't see any axes at all except the left stick (the problem with GTA3 is that it uses the trigger axes as right stick and doesn't allow you to change or even disable that mapping, which makes the game unplayable).

Unfortunately it seems that MAMEOSX also doesn't play nice with key injection apps (I tried both Enjoyable and Joystick Mapper). Having quickly glanced at the source code I suspect this is because it's reading the keyboard at a lower level than the key events get injected, so adding key injection to this driver wouldn't help you anyway. AFAIK the only way to inject events at that level would be to implement a virtual HID device driver (e.g. https://github.com/unbit/foohid) and that is most likely way outside the scope of this project.

So part from someone fixing MAMEOSX itself I think you're out of luck. It definitely is that which is doing something wrong with joystick input because [the Mac SDL port of MAME 0.135](http://sdlmame.lngn.net/stable/) does recognize the right stick axes. It's just bare command line MAME though, without the nice interface and other additions, but if you want to try it;

1. Download sdlmame0135-x86_64.tar.bz2 and extract it somewhere
2. Download SDL-1.2.15.dmg from [here](https://www.libsdl.org/download-1.2.php). Create a directory called Frameworks in the same place as the sdlmame directory and copy SDL.framework from the dmg into that folder.
3. Run it with the `-video opengl` switch because software rendering doesn't work.

daxfox, 2018-11-14:
Yes it’s MAME 0.135… Cool, thanks for confirming what I suspected and for providing a possible workaround with the SDL port.

closed 2019-05-26 by FranticRain