360Controller/360Controller #711: hori fighting edge x360 wired joystick not part of driver

Pogo9, 2018-07-02:
### Type of Controller
x360 wired joystick

### OS Version
10.12.6

### Driver Version
16.8

### Connection Method
wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID:0x5503
<br/>Vendor ID:0x24c6

### Response of Controller
the stick appears in the list of usb devices connected and detected by osx, but does not appear as a controller in the pane

### Response in Preference Pane
does not appear as a select-able controller in the pane

### Detailed Explaination of Issue
checked the kext to see if the product and vendor entry is there.  other Hori products appear, but not this product and vendor combo

trying desperately to get this joystick working in osx.  please help!

FranticRain, 2018-07-08:
Here is a build with this controller added.

[Install360Controller.pkg.zip](https://github.com/360Controller/360Controller/files/2174128/Install360Controller.pkg.zip)

closed 2018-07-08 by FranticRain

Pogo9, 2018-07-09:
Thank you!  works perfectly!!!!!!