360Controller/360Controller #713: MadCatz Beat Pad not part of driver

jrab66, 2018-07-04:
### Type of Controller
MadCatz Beat Controller

### OS Version
10.11.4

### Driver Version
0.16.8

### Connection Method
Wired
### Device Name and Info
MadCatz Beat Pad
>Device Name:
<br/>Product ID: 0x4740
<br/>Vendor ID: 0x0738

### Response of Controller
the yellow light only power on 1 time when connected

(already check in linux and is recognised without problem as madcatz beat pad)

### Response in Preference Pane
does not apear

### Detailed Explaination of Issue
it seems that not work the driver in this peripherial, is not detected as joystick,

jrab66, 2018-07-04:
Already do a push request  i  add the device in the Info.plist

closed 2018-07-05 by jrab66

FranticRain, 2018-07-08:
Here's a build with this controller added.

[Install360Controller.pkg.zip](https://github.com/360Controller/360Controller/files/2174129/Install360Controller.pkg.zip)

jrab66, 2018-07-16:
thanks lml , i dont have the dependancy in my machine lml