360Controller/360Controller #715: XBOX One controller is not visible to 360Controller while in bluetooth mode

vergen, 2018-07-09:
XBOX One controller

OSX: 10.13.5

360Controller: 0.16.8

Bluetooth

<br/>Product ID:0x02FD
<br/>Vendor ID:0x045E

It successfully connects to my mac via bluetooth with no issues.

It only says no devices found. When I plug it in with USB it is instantly detected. Although I've never been able to get force feedback to work even plugged in.

It connects to my computer like it should but the app is unable to see my controller and I am unable to get force feedback. If I use USB the controller is seen by the app and is configurable.

vergen, 2018-07-09:
The driver is also disabled in the app every time I restart my computer. The computer also blanks to a black screen which forces me to do a hard reset, this happened when I installed the driver but I am unsure of why it's happening.

FranticRain, 2018-07-11:
Regarding Bluetooth: https://github.com/360Controller/360Controller#xbox-one-controllers-connected-with-bluetooth
You don't need the driver when using the controller via Bluetooth, so we don't support it. It won't show up in the preference pane. That is just work wired and wireless dongle controllers. Supporting the controller in Bluetooth mode is the game developer's problem.

Regarding black screen: You are possibly experiencing a kernel panic? Not sure about that one. Try unplugging your controllers before a shut down, just to make sure.

Regarding USB: If it works in the preference pane, then it's working and that's all we can do. The game has to correctly request force feedback. That's also the game developer's problem.

vergen, 2018-07-11:
Yeah I'll just install the driver for now. When I login it takes way longer than before also so I suspect this is a driver issue.

I thought the driver enabled force feedback when in Bluetooth mode, I must have misunderstood that part then.

FranticRain, 2018-07-12:
The driver was supposed to enable force feedback on a Bluetooth controller, and in some cases does, but implementation was abandoned so that line is sort of an ill-informed reminder of that failed effort. It could probably afford to be changed, but it is still technically correct for SDL games.

closed 2018-07-12 by FranticRain