360Controller/360Controller #716: Xbox one S controller not showing up, whether it is connected via usb or bluetooth

Liamm2016, 2018-07-11:
### Type of Controller
Xbox One S

### OS Version
MacOS sierra 10.12.3

### Driver Version
Latest version 0.16.8

### Connection Method
Wired and Bluetooth

### Device Name and Info
>Device Name: Unsure
<br/>Product ID: 0x02FD
<br/>Vendor ID: 0x045E
I believe that this is the right info, this is from the system information page.

### Response of Controller
It vibrates once plugged in

### Response in Preference Pane
Unsure

### Detailed Explanation of Issue
When plugged into my mac the controller vibrates, but does not show up as a controller. When connected by bluetooth it doesn't show up either

FranticRain, 2018-07-12:
I'm extremely confused by your issue, so let's try and get the basics here. In `Apple Menu` -> `System Preferences`, after you install the driver, there is a new option called something like `Xbox 360 Controller`. This is the "Preference Pane." This preference pane is likely telling you "No device found." If it has found a device, then the driver is working and your issue is specific to a game, not to the driver. My bet is that it isn't working, because you didn't say that the controller lights up, like it should. So let's try a little debugging.

I need you to open up your terminal and run `sudo kextload /Library/Extensions/360Controller.kext` and tell me what it says.

If it says `/Library/Extensions/360Controller.kext failed to load - (libkern/kext) system policy prevents loading; check the system/kernel logs for errors or try kextutil(8).` Then you'll need to allow the driver in the `Security & Privacy` section of your `System Preferences`.
[image]

Otherwise, this issue will be closed down, because we do not support counterfeit controllers.

ModernDoughnutz, 2018-07-21:
Same is happening to me. I have an Xbox One S controller and I have it connected via micro-usb. It vibrates once it connects so I know it's not the cable. My controller doesn't turn on and when it turns on it tries to connect to my bluetooth. ( Oh by the way using the bluetooth doesn't work either but I'm not sure if it's supposed to ) I typed in the thing to the Terminal and it says something about the "local System Administrator"

FranticRain, 2018-07-22:
The controller vibrating is not an indicator of a good connection. That vibration is simply how the controller responds to power input through the USB connection. You could literally plug the controller into a DC wall adapter like you use for a phone and the controller will vibrate. There are two types of micro USB cables. "Charge" and "data" cables. Charge cables only have power connections, while data cables have power and data connections. The only way to know for sure if the cable is working correctly is to follow the debugging tips listed in the README to see if your controller shows up properly in the system report.

The reason your controller tries to connect when you turn it on is because it thinks it is in a wireless mode, but being charged via a cable. It hasn't received the magic USB startup packet. Starting the controller with the guide button always starts wireless mode.

Bluetooth does work. It doesn't connect to the driver, as documented in the README. The best way to test it would be in the browser with some sort of gamepad tester. I use [this website](http://html5gamepad.com/) a lot.

As for the response when you tried `kextload`, I'm going to need you to be more specific with the details. Ideally a copy paste of the output. It's hard for me to work with vague error messages.

Furthermore, if you could please open a separate issue for your problem, that would be appreciated. People actually rarely have identical issues to others and often bandwagon on a fairly generic issue title and we end up having to juggle five different issues in one thread. It's a huge hassle and makes it hard to track issues.

manni1234, 2018-09-21:
When i connect it it works with the bluetooth but it says no device found under Xbox 360 Controllers

FranticRain, 2018-09-22:
@manni1234 Bluetooth controllers are not supported by this driver. They will not show up in the preference pane of this driver. Bluetooth Xbox One controllers are supported by default in the operating system, and therefore do not need a driver to function. You only need the driver if you wish to connect your Xbox One controller via USB. If you would like to test if your Xbox One Bluetooth controller is properly connected to your computer, I recommend a HTML5 gamepad tester like [this one.](http://html5gamepad.com/)

FranticRain, 2019-01-15:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-01-15 by FranticRain