360Controller/360Controller #717: Guitar Hero Live controller: is there a way to make it work?

mikkabouzu, 2018-07-14:
### Type of Controller
Guitar Hero live guitar with Xbox 360 receiver (green button)

### OS Version
10.13.4

### Driver Version
0.16.8 (tried 0.16.5 and 0.16.4 too)

### Connection Method
wireless receiver

### Device Name and Info
>Device Name: Guitar Hero
>Product ID: 0x070b
>Vendor ID: 0x1430  (RedOctane)
>Version:	1.16
>Serial Number:	B8AA857F
>Speed:	Up to 12 Mb/sec
>Manufacturer:	ACtivision
>Location ID:	0xfa130000 / 5
>Current Available (mA):	500
>Current Required (mA):	100
>Extra Operating Current (mA):	0

### Response of Controller
A added the device info to `WirelessGamingReceiver.kext` (based on `WirelessGamingReceiverForWindows`)  and `360Controller.kext` (based on the `Controller` device).
After this when pressing the connect button twice on the guitar, the four leds blink rapidly, and the receiver's led starts to blink rapidly, but nothing else happens.

The receiver and the controller were tested on Windows and are working flawlessly.

### Response in Preference Pane
It appears as Xbox 360 controller (Wired), nothing is registering.

### Detailed Explanation of Issue
I'm trying to figure out if it's possible to make this work the Guitar Hero Live controller (with the xbox 360 receiver).

jrab66, 2018-07-16:
already add the description for this device! :+1: 

commit #718

FranticRain, 2018-08-06:
I'll merge and build this device, but I believe that the packet format of guitar controllers isn't supported for Xbox 360 controller types. Someone will need to implement that, and I'm certainly not volunteering.

[Install360Controller.pkg.zip](https://github.com/360Controller/360Controller/files/2260958/Install360Controller.pkg.zip)

closed 2018-08-06 by FranticRain