360Controller/360Controller #719: Installation stuck "registering updated components"

FallOfArca, 2018-07-18:
### Type of Controller
n/a

### OS Version
10.13.5

### Driver Version
0.16.8

### Connection Method
n/a

### Device Name and Info
>Device Name: n/a
<br/>Product ID: n/a
<br/>Vendor ID: n/a

### Response of Controller
n/a

### Response in Preference Pane
n/a

### Detailed Explaination of Issue
When I tried to install, my laptop ran out of power. When I powered it back up I had to restart the installation, but now every time I try, it stalls at "registering updated components" and says there's less than a minute left. When I go into System Preferences, the pane does show up even though the install isn't technically complete, but it doesn't seem to work. I've tried uninstalling via the preferences pane and reinstalling, but it still gets stuck.

mrMagicCod3r, 2018-07-27:
I got the same during uninterrupted update from previous version on my mac pro. How do I uninstall this thing?

jonnymorrisuk, 2018-12-03:
Same here, in my case when I installed it first time the installation got as far as requiring permission for the developer but it completed before I could click 'Allow' in the Security and Privacy preference pane; the XBox One controller was not recognised so I tried to re-install but it has always got stuck at this same point ('registering updated components'.  I tried deleting all files installed (as listed by the installer - 'list files') but to no avail.

dkliman, 2020-12-14:
I have the same problem. did anybody solve it?

jonnymorrisuk, 2020-12-14:
Honestly I can't remember exactly why I wanted to use an XBox controller on my Mac, I think it might have been for Steam remote play.  Suffice to say I don't think I ever got it working as I don't use any game controllers through my Mac, and I gave up on Steam remote play probably for that reason, and the trackpad is terrible for PC games that require a mouse.  Sorry I can't be of any more help.