360Controller/360Controller #720: /Library/Extensions/360Controller.kext failed to load - com.mice.driver.Xbox360Controller's dependencies failed security checks; failing (detailed logs attached)

Nowaker, 2018-07-19:
### Type of Controller

Xbox 360 (USB)

### OS Version

High Sierra 10.13.5

### Driver Version

0.16.8

### Connection Method

Wired

### Device Name and Info

>Device Name:
<br/>Product ID: 0x028e
<br/>Vendor ID: 0x045e

### Response of Controller

None

### Response in Preference Pane

None, not detected

### Detailed Explaination of Issue

```
% sudo kextload /Library/StagedExtensions/Library/Extensions/360Controller.kext
/Library/StagedExtensions/Library/Extensions/360Controller.kext failed to load - (libkern/kext) dependency load failed; check the system/kernel logs for errors or try kextutil(8).
```

When trying to enable the checkbox "Enable Driver" in the applet:

```
% tail -f /var/log/system.log
(...)
Jul 18 21:48:20 nwkr-mbp System Preferences[865]: APPLESCRIPT ERROR:
	{
	    NSAppleScriptErrorAppName = "System Preferences";
	    NSAppleScriptErrorBriefMessage = "/Library/Extensions/360Controller.kext failed to load - (libkern/kext) dependency load failed; check the system/kernel logs for errors or try kextutil(8).";
	    NSAppleScriptErrorMessage = "/Library/Extensions/360Controller.kext failed to load - (libkern/kext) dependency load failed; check the system/kernel logs for errors or try kextutil(8).";
	    NSAppleScriptErrorNumber = 71;
	    NSAppleScriptErrorRange = "NSRange: {0, 143}";
	}
	ERROR LOCATION:
	***ERROR_HERE--->do shell script "            cd \"/Library/Extensions\"
	            kextload \"360Controller.kext\"
	            " with administrator privileges<---***
```

Output of Console.app when trying to load the driver:

```
default	21:50:21.098794 -0500	kextd	/Library/StagedExtensions/Library/Extensions/360Controller.kext - no compatible dependency found for com.apple.iokit.IOUSBFamily.
default	21:50:21.104677 -0500	trustd	asynchronously fetching CRL (http://crl.apple.com/codesigning.crl) for client (kextd[62]/0#-1 LF=0)
default	21:50:21.314269 -0500	kextd	com.mice.driver.Xbox360Controller's dependencies failed security checks; failing.
```

Please let me know if you need anything else. Please note I did authorize your extension in "Security & Privacy" tab upon installation (multiple times, after uninstall-reboot-install cycle).

Nowaker, 2018-07-19:
`csrutil disable` in Recovery helped with this anomaly. That said, (2) is still valid getting the UI more informative would be helpful.

1. ~~~The kernel extension doesn't load.~~~
2. The configuration applet ignored non-zero exit status for kextload and "thinks" it enabled the extension while in fact it did not. An error message should be provided instead. Best if the applet can collect debug information related to the failure.

FranticRain, 2018-07-22:
2 is a known issue. If you fix it, I'd be more than happy to accept a pull request. [The code you are looking for is right here.](https://github.com/360Controller/360Controller/blob/dd1ce4a5a4685c3825a231a571f3c31342da320e/Pref360Control/Pref360ControlPref.m#L1054) You'll want to make sure that you follow the new best practices for kext loading outlined in #610.

FranticRain, 2018-12-04:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-12-04 by FranticRain