360Controller/360Controller #722: Buttons aren't responding

warbison, 2018-07-27:
### Type of Controller
Razer wolverine ultimate

### OS Version
10.12.6

### Driver Version
0.16.8

### Connection Method
Wired 

### Device Name and Info
>Device Name:
<br/>Product ID:0x0a14
<br/>Vendor ID:0x1532

### Response of Controller
Lights up and vibrates

### Response in Preference Pane
Appears in preference pane but the buttons don't respond

### Detailed Explaination of Issue
Controller turns on but none of the buttons are responding through the software. (in the console) When I plug in the controller I get this: default	03:10:39.461661 -0600	kernel	Razer Wolverine Ultimate: family specific matching fails
default	03:10:39.461679 -0600	kernel	Razer Wolverine Ultimate: family specific matching fails
default	03:10:39.461685 -0600	kernel	Razer Wolverine Ultimate: family specific matching fails
default	03:10:39.461793 -0600	kernel	Razer Wolverine Ultimate: family specific matching fails
default	03:10:39.461859 -0600	kernel	Razer Wolverine Ultimate: family specific matching fails
default	03:10:39.461934 -0600	kernel	Razer Wolverine Ultimate: family specific matching fails
default	03:10:39.462026 -0600	kernel	Razer Wolverine Ultimate: family specific matching fails
default	03:10:39.462757 -0600	kernel	Razer Wolverine Ultimate: family specific matching fails
default	03:10:39.462861 -0600	kernel	Razer Wolverine Ultimate: family specific matching fails
default	03:10:39.462868 -0600	kernel	Razer Wolverine Ultimate: family specific matching fails
default	03:10:39.462872 -0600	kernel	Razer Wolverine Ultimate: family specific matching fails
default	03:10:39.462909 -0600	kernel	Xbox360Peripheral::probe(Razer Wolverine Ultimate)
default	03:10:39.462916 -0600	kernel	Xbox360Peripheral::start(Razer Wolverine Ultimate) <1>
default	03:10:39.531031 -0600	kernel	Unknown device release 0101
default	03:10:39.531602 -0600	kernel	start - unable to find chatpad interface
default	03:10:39.538423 -0600	kernel	Unknown escape 128
default	03:10:56.014741 -0600	kernel	SmartBattery: finished polling type 4

FranticRain, 2018-08-06:
This controller likely uses a custom button packet. You're going to need to sniff the packets using a Windows PC before it can be supported. [Here are the steps to complete this.](https://vestigl.com/sniffing/index.html)

FranticRain, 2018-12-04:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-12-04 by FranticRain