360Controller/360Controller #724: Xbox One Thrustmaster TMX Wheel and Pedals not recognized my 360 Controller

PavementPilot, 2018-07-30:
### Type of Controller
Xbox One

### OS Version
10.13.6

### Driver Version
0.16.8

### Connection Method
Wired

### Device Name and Info
Thrustmaster TMX GIP Racing Wheel:

  Product ID:	0xb67e
  Vendor ID:	0x044f  (ThrustMaster, Inc.)
  Version:	1.01
  Serial Number:	000081F6E78C4337
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Thrustmaster
  Location ID:	0x14100000 / 13
  Current Available (mA):	500
  Extra Operating Current (mA):	0

### Response of Controller
When plugged into the Mac, it powers on and the wheel does a full left and right test rotation. the white power light is on, and the mode light is red.

### Response in Preference Pane
No device found

### Detailed Explaination of Issue
As explained in Response of Controller, the 360 Controller does not recognize the Thrustermaster TMX wheel or pedals.

PavementPilot, 2018-08-03:
I see that Issue 725 got fast action for adding support, but my issue 724 is still sitting without help. What gives?

FranticRain, 2018-08-06:
[Install360Controller.pkg.zip](https://github.com/360Controller/360Controller/files/2260948/Install360Controller.pkg.zip)

closed 2018-08-06 by FranticRain

PavementPilot, 2018-08-06:
Thanks for getting to it. Works good now. With the amount of buttons available on this device, there are limitations with using it for Euro Truck Sim 2 or American Truck Sim. Some of the buttons double up on the program, but are fine in the 360Controller. I am satisfied with what I have though.