360Controller/360Controller #727: Program failing to recognise wired Xbox One S

utopia100, 2018-08-06:
### Type of Controller
Xbox One S

### OS Version
High Sierra v10.13.5 (17F77)

### Driver Version
Version 0.16.8

### Connection Method
Wired

### Device Name and Info
Not sure what readme is being referred to, so if you could link it in a reply that'd be great.

### Response of Controller
Controller vibrates in response to it receiving power.

### Response in Preference Pane
None

### Detailed Explaination of Issue
I'm able to get my controller to work in bluetooth mode, but upon attempting to connect via wired the 360Controller program doesn't appear to recognise it. As a side note, the "Enable driver" checkbox was unticked by default, and unticks itself every time I close system preferences. I'm guessing this must be the cause of the issue because the driver doesn't seem to be enabling itself.

FranticRain, 2018-08-08:
The thing referred to by the "Device Name and Info" section is this: https://github.com/360Controller/360Controller#how-do-i-find-my-vendor-id-and-product-id

When connected via Bluetooth, the preference pane will never recognize the controller, because it isn't using the driver. The preference pane is only recognizes controllers that are using the driver.

1400k, 2018-08-22:
Wait, so why is my 'Enable Driver' button off? Whenever I plug in the Xbox One controller (not third party) it just gives a little rumble to say "I have power" but it will not light up, and the emulator will not read it even after I have enabled the driver button. Also, whenever I leave the emulator, the 'Enable Driver' button is disabled.

FranticRain, 2018-08-23:
Here is the code that governs that checkbox. For reference, the checkbox is unchecked by default.
https://github.com/360Controller/360Controller/blob/9cae07189fd7e5bd4eff7a24ba0e4f6530cce873/Pref360Control/Pref360ControlPref.m#L849-L856
So when you say your controller doesn't show up in the preference pane, and the checkbox is unchecked, that's because:

1. The driver hasn't been loaded, because it hasn't found a valid match.
2. Because that check isn't made unless the preference pane recognizes a connected controller.

The enable checkbox status is completely arbitrary, generally bad, and once again __not an indicator of an issue.__ Now, if you would like to get some help and potentially resolve your issue, please create your own issue instead of commenting on an pre-existing issue.

FranticRain, 2018-12-04:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-12-04 by FranticRain