360Controller/360Controller #728: Pedals of the Thrustmaster TMX need reassigning

PavementPilot, 2018-08-08:
Type of Controller
Xbox One

OS Version
Mac 10.13.6

Driver Version
0.16.8

Connection Method
Wired

Device Name and Info
Thrustmaster TMX GIP Racing Wheel:

Product ID:	0xb67e
Vendor ID:	0x044f (ThrustMaster, Inc.)
Version:	1.01
Serial Number:	000081F6E78C4337
Speed:	Up to 12 Mb/sec
Manufacturer:	Thrustmaster
Location ID:	0x14100000 / 13
Current Available (mA):	500
Extra Operating Current (mA):	0

Response of Controller
Pedals are linked to the triggers. They need to be linked to the right joystick. 

Response in Preference Pane
When the gas pedal is pressed, the right trigger indicates input, but because of the range of input, the trigger goes full then null then full, multiple times, until the pedal is fully pressed. same goes for brake pedal on the left trigger.

Detailed Explaination of Issue
If linked to the right joystick axis y, then the full range of input would be correct.

PavementPilot, 2018-08-10:
On further thought, I'm not sure if reassigning the pedals will work. it may be how the driver is reading the pedals. It may be that the triggers have a value of say 0 to 50, and the pedals at full travel have a value from 0 to say 200. I notice that the triggers register the pedal from null to full 4 times in the pedals entire travel.

FranticRain, 2018-08-11:
How's this work?
[Install360Controller.pkg.zip](https://github.com/360Controller/360Controller/files/2280369/Install360Controller.pkg.zip)

PavementPilot, 2018-08-11:
Perfect. Full proper travel gives full proper registration on the driver. Thank you.

closed 2018-08-11 by FranticRain