360Controller/360Controller #729: Xbox 360 Controller Panel Bug: Advanced Settings issue

Jakedjoe, 2018-08-13:
### Type of Controller
Xbox 360: Third party, PowerA

### OS Version
macOS 10.13.6

### Driver Version
Version 0.16.8

### Connection Method
Wired

### Detailed Explaination of Issue
Every time I start my game (Fortnite) the controller's mapping is all off. In order for it to work I have to launch the Xbox 360 Controller Panel and go to the Advanced tab. The tab remembers my previous settings, but I need to; Uncheck Enable Driver, Recheck Enable Driver, then I uncheck Pretend to be an Xbox 360 Controller and then recheck Pretend to be an Xbox 360 Controller. After that the driver works fine and I can play my game. 

It seems that only the interface check box status is remembered, and not the actual settings. 

I had a similar problem before where it would not remember the Pretend to be an Xbox 360 Controller setting, and I would only have to check the Pretend to be an Xbox 360 Controller box. Now it remembers that the box is checked but the setting does not hold in the driver.

FranticRain, 2018-08-14:
This sounds like the settings are being saved, but aren't properly being loaded into the controller at time of start. Do you plug your controller in after boot, or is it plugged in when you start the computer? Does this occur after a boot every time, or can you recreate this by just unplugging and then reconnecting the controller without a reboot? I'm not sure if there is much we can do about this one because it would be an issue with the kernel space driver not having access to settings during start up. It would go along with the settings rewrite task, however unlikely that is to happen.

Jakedjoe, 2018-08-14:
I can recreate this problem by just unplugging and then reconnecting the controller without a reboot. I need to use "Pretend to be an Xbox 360 Controller" or the mapping would be off. But simple resetting "Pretend to be an Xbox 360 Controller" is not enough, I need to disable the driver, then re-enable it.

FranticRain, 2018-08-14:
Is the mapping off in the preference pane or in a specific game? For some reason, you aren't seeing hot changes in the settings, and that's very peculiar.

Jakedjoe, 2018-08-15:
The mapping is off only for specific game (fortnite). When "Pretend to be an Xbox 360 Controller" is OFF the mapping in preference pane works fine, and does not work with the game. When "Pretend to be an Xbox 360 Controller" is ON the mapping in preference pane does not register anything, however it works fine with the game. 

Another thing I noticed. When I restarted my Mac the "Enable Driver" was toggled OFF.

IAmOrion, 2020-04-29:
April 29th 2020 and still having these same issues - Fortnite only works when "Pretend to be a 360 Controller" is selected, then, if controller is disconnected for any reason, even momentarily, have to go back to settings, deselect then reselect before it will work again in Fortnite