360Controller/360Controller #730: Macbook broken after installing .dmg file

vmunv, 2018-08-14:
After my computer restarted for the software, I can log in to my user profile but the bar simply loads all the way but never actually logs in (I waited an hour for it to do something but nothing ever happened).... I'm got high sierra so I'm pretty sure I meet the system requirements for the download....

actionadam90, 2018-08-17:
On one of my Macs it worked any my computer was fine but on another it broke

vmunv, 2018-08-17:
so apparently the drive became unmounted or something so i just had to remount and then everything worked fine

bpcatsolton, 2018-08-26:
Hi !
My MacBook Pro 2018 also has the same problem for the boot after installing the drivers ;-(
How do you manage to "remount the drive", please ?
What kind of commands ?

ahokinson, 2018-09-09:
I apologize for not seeing this earlier. I was able to safe boot and believe it or not @bpcatsolton's suggestion to reselect the boot drive actually worked... I have since completely removed the package after getting back into my system.

FranticRain, 2018-09-15:
This should be resolved in [0.16.9](https://github.com/360Controller/360Controller/releases/tag/v0.16.9). Further discussion has been moved to #698.

closed 2018-09-15 by FranticRain