360Controller/360Controller #735: Controller buttons stop responding

LauraThompson, 2018-08-25:
### Type of Controller
Wired XBox 360 controller

### OS Version
Mac OS High Sierra 10.13.6

### Driver Version
Tried both 15(beta) and 16.8

### Connection Method
Wired

### Device Name and Info
Product ID:	0x028e
Vendor ID:	0x045e  (Microsoft Corporation)

### Response of Controller
The lights come on the controller. 
Controller indicates first player. 

### Response in Preference Pane
Mac can see controller, the controller button test diagram on the screen indicates that the buttons work for up to twenty seconds but then stop responding, sometimes in a held state.

### Detailed Explaination of Issue
Controller stops responding after 2-20 seconds of being plugged in (controller works fine in a windows laptop). 

### Output from console app while plugging in and testing buttons on controller:
 
default	20:28:24.283751 +0100	icdd	#ICDebug - 37:{ICWiredBrowser.m} (USB Interface first match)
default	20:28:24.284513 +0100	icdd	#ICDebug - 388:{ICWiredBrowser.m} (7 USB Descriptions Managed)
default	20:28:24.285676 +0100	icdd	#ICDebug - 457:{ICDDMessageCenter.m} (+Add Controller - 0xff/0x5d/0x1 - 0xfa130000 - ICDeviceDescriptionSUQuery)
default	20:28:24.285744 +0100	icdd	#ICDebug - 205:{ICResourceManager.m} (00000000-0000-0000-0000-0000045E028E|Controller|(null)|SW=FALSE|)
default	20:28:24.286594 +0100	icdd	#ICDebug - 37:{ICWiredBrowser.m} (USB Interface first match)
default	20:28:24.287192 +0100	icdd	#ICDebug - 457:{ICDDMessageCenter.m} (+Add Controller - 0xff/0x5d/0x1 - 0xfa130000 - ICDeviceDescriptionAdded)
default	20:28:24.287339 +0100	icdd	#ICDebug - 388:{ICWiredBrowser.m} (8 USB Descriptions Managed)
default	20:28:24.288514 +0100	icdd	#ICDebug - 37:{ICWiredBrowser.m} (USB Interface first match)
default	20:28:24.289040 +0100	icdd	#ICDebug - 457:{ICDDMessageCenter.m} (+Add Controller - 0xff/0x5d/0x3 - 0xfa130000 - ICDeviceDescriptionSUQuery)
default	20:28:24.289534 +0100	icdd	#ICDebug - 205:{ICResourceManager.m} (00000000-0000-0000-0000-0000045E028E|Controller|(null)|SW=FALSE|)
default	20:28:24.289917 +0100	icdd	#ICDebug - 388:{ICWiredBrowser.m} (9 USB Descriptions Managed)
default	20:28:24.290756 +0100	icdd	#ICDebug - 457:{ICDDMessageCenter.m} (+Add Controller - 0xff/0x5d/0x3 - 0xfa130000 - ICDeviceDescriptionUndefined)
default	20:28:24.290864 +0100	icdd	#ICDebug - 37:{ICWiredBrowser.m} (USB Interface first match)
default	20:28:24.291517 +0100	icdd	#ICDebug - 388:{ICWiredBrowser.m} (10 USB Descriptions Managed)
default	20:28:24.292089 +0100	icdd	#ICDebug - 457:{ICDDMessageCenter.m} (+Add Controller - 0xff/0x5d/0x2 - 0xfa130000 - ICDeviceDescriptionSUQuery)
default	20:28:24.292107 +0100	icdd	#ICDebug - 205:{ICResourceManager.m} (00000000-0000-0000-0000-0000045E028E|Controller|(null)|SW=FALSE|)
default	20:28:24.293295 +0100	icdd	#ICDebug - 457:{ICDDMessageCenter.m} (+Add Controller - 0xff/0xfd/0x13 - 0xfa130000 - ICDeviceDescriptionSUQuery)
default	20:28:24.293365 +0100	icdd	#ICDebug - 205:{ICResourceManager.m} (00000000-0000-0000-0000-0000045E028E|Controller|(null)|SW=FALSE|)
default	20:28:24.294463 +0100	icdd	#ICDebug - 457:{ICDDMessageCenter.m} (+Add Controller - 0xff/0x5d/0x2 - 0xfa130000 - ICDeviceDescriptionUndefined)
default	20:28:24.295588 +0100	icdd	#ICDebug - 457:{ICDDMessageCenter.m} (+Add Controller - 0xff/0xfd/0x13 - 0xfa130000 - ICDeviceDescriptionUndefined)
default	20:28:24.303153 +0100	loginwindow	IOHIDDeviceClass::start: elementCount=22 reportHandlerCount=1
default	20:28:24.303202 +0100	loginwindow	IOHIDDeviceClass::buildElements: type=0 *buffer=7fee33062c30 *count=22 size=2112
default	20:28:24.303241 +0100	loginwindow	IOHIDDeviceClass::buildElements: type=1 *buffer=60c00012bd90 *count=1 size=96
default	20:28:24.305199 +0100	360Daemon	IOHIDDeviceClass::start: elementCount=22 reportHandlerCount=1
default	20:28:24.305291 +0100	360Daemon	IOHIDDeviceClass::buildElements: type=0 *buffer=7fd159003030 *count=22 size=2112
default	20:28:24.305355 +0100	360Daemon	IOHIDDeviceClass::buildElements: type=1 *buffer=7fd157f05530 *count=1 size=96
default	20:28:24.305943 +0100	icdd	#ICDebug - 23:{ICWiredBrowser.m} (USB Device first match)
default	20:28:24.306495 +0100	icdd	#ICDebug - 388:{ICWiredBrowser.m} (11 USB Descriptions Managed)
default	20:28:24.306496 +0100	System Preferences	IOHIDDeviceClass::start: elementCount=22 reportHandlerCount=1
default	20:28:24.306542 +0100	System Preferences	IOHIDDeviceClass::buildElements: type=0 *buffer=7fe0fb175030 *count=22 size=2112
default	20:28:24.306582 +0100	System Preferences	IOHIDDeviceClass::buildElements: type=1 *buffer=6000001354d0 *count=1 size=96
default	20:28:24.307238 +0100	icdd	#ICDebug - 457:{ICDDMessageCenter.m} (+Add Controller - 0x0/0x0/0x0 - 0xfa130000 - ICDeviceDescriptionSUQuery)
default	20:28:24.308400 +0100	System Preferences	IOHIDDeviceClass::start: elementCount=22 reportHandlerCount=1
default	20:28:24.308448 +0100	System Preferences	IOHIDDeviceClass::buildElements: type=0 *buffer=7fe0fc074e30 *count=22 size=2112
default	20:28:24.308508 +0100	System Preferences	IOHIDDeviceClass::buildElements: type=1 *buffer=60c000134170 *count=1 size=96
default	20:28:24.259776 +0100	kernel	000308.090243 Controller@fa130000: IOUSBHostDevice::getDescriptorGated: compliance violation: USB 2.0 9.3.5: device returned more than wLength data
default	20:28:24.264523 +0100	kernel	Controller: family specific matching fails
default	20:28:24.264530 +0100	kernel	Controller: family specific matching fails
default	20:28:24.264536 +0100	kernel	Controller: family specific matching fails
default	20:28:24.264542 +0100	kernel	Controller: family specific matching fails
default	20:28:24.264546 +0100	kernel	Controller: family specific matching fails
default	20:28:24.264551 +0100	kernel	Controller: family specific matching fails
[... 46 lines cut ...]
default	20:28:24.264821 +0100	kernel	Xbox360Peripheral::probe(Controller)
default	20:28:24.264827 +0100	kernel	Xbox360Peripheral::start(Controller) <1>
default	20:28:24.301860 +0100	kernel	start - failed to read chatpad setting (e000404f): 0xe000404f (UNDEFINED)
default	20:28:24.369645 +0100	icdd	#ICDebug - 205:{ICResourceManager.m} (00000000-0000-0000-0030-423338353534|Controller|MANUFACTURER:Holtek;MODEL:Controller|SW=FALSE|)
default	20:28:24.370353 +0100	icdd	#ICDebug - 457:{ICDDMessageCenter.m} (+Add Controller - 0x0/0x0/0x0 - 0xfa130000 - ICDeviceDescriptionUndefined)
error	20:28:32.270832 +0100	WindowServer	[ERROR] - Unknown CGXDisplayDevice: 0x41dc9d00
error	20:28:32.271907 +0100	WindowServer	[ERROR] - Unknown CGXDisplayDevice: 0x41dc9d00
error	20:28:32.371552 +0100	WindowServer	[ERROR] - Unknown CGXDisplayDevice: 0x41dc9d00
error	20:28:32.372604 +0100	WindowServer	[ERROR] - Unknown CGXDisplayDevice: 0x41dc9d00
error	20:28:32.956426 +0100	WindowServer	[ERROR] - Unknown CGXDisplayDevice: 0x41dc9d00
error	20:28:32.957517 +0100	WindowServer	[ERROR] - Unknown CGXDisplayDevice: 0x41dc9d00
default	20:28:32.974552 +0100	pboard	Initalizing PBIR Converters
default	20:28:32.977804 +0100	pboard	Registering converter for type: public.url -> UAPBIRPublicURLConverter
default	20:28:33.005496 +0100	pboard	Sending pasteboard update to server: <private>
default	20:28:33.005852 +0100	pboard	[Local Pasteboard] Adding alias: public.text for type: public.utf8-plain-text
default	20:28:33.015637 +0100	useractivityd	[PBOARD CONTROLLER] Received new client connection: <private>
default	20:28:33.016589 +0100	useractivityd	[PBOARD CONTROLLER] Got type update from client, gen: 1, <private>
default	20:28:33.018882 +0100	useractivityd	DETERMINE: 14C04569-DC89-464A-B1CA-96681F9D605F/<private> as the current item-to-advertise.
default	20:28:33.057329 +0100	pboard	Sending pasteboard update to server: <private>
default	20:28:33.057442 +0100	pboard	[Local Pasteboard] Adding alias: public.text for type: public.utf8-plain-text
error	20:28:33.057904 +0100	WindowServer	[ERROR] - Unknown CGXDisplayDevice: 0x41dc9d00
error	20:28:33.058906 +0100	WindowServer	[ERROR] - Unknown CGXDisplayDevice: 0x41dc9d00
default	20:28:33.061820 +0100	useractivityd	[PBOARD CONTROLLER] Got type update from client, gen: 1, <private>
default	20:28:33.063609 +0100	useractivityd	DETERMINE: 14C04569-DC89-464A-B1CA-96681F9D605F/<private> as the current item-to-advertise.
default	20:28:34.567452 +0100	trustd	could not enable test hierarchy: no UAT pinning preferences set

FranticRain, 2018-08-29:
When the controller stops responding/hangs, if you back out of the preference pane and return to it, does the preference pane still think that the controller is connected? I need to figure out if this is a kext load issue, or some sort of operational issue first. If you do test it again, don't press the "Guide"/"Xbox" button on the controller and see if it changes the results at all.

The logs imply that the driver sees a chatpad connected to the controller? Is there a chatpad connected to the controller? And if so, can you try using the driver without the chatpad?

LauraThompson, 2018-09-10:
If it works for a few seconds without me pressing the “x” button, then hangs, the mac still sees it’s connected when I back out and return to the preference pane.
There is no chat pad or anything connected to the controller...

LauraThompson, 2018-09-12:
I've tried downloading it a couple of times but it just hangs with less than a minute left and I have to restart my mac because force quitting the program doesn't work.

[image]

FranticRain, 2018-09-14:
I'm not quite sure what to tell you. I just ran it myself and it works just fine. Failing to install a .pkg would be an Apple problem, not a me problem. That's their code.

FranticRain, 2019-01-15:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-01-15 by FranticRain