360Controller/360Controller #736: "No Device Found" In Preference Pane despite controller being recognised via System Report > USB

georgeawhite, 2018-08-28:
### Type of Controller
Xbox One

### OS Version
High Sierra 10.13.6

### Driver Version
0.16.8
### Connection Method
Wired

### Device Name and Info
>Device Name:
<br/>Product ID: 0x02dd
<br/>Vendor ID: 0x045e 

### Response of Controller
When plugging in the controller via micro usb it vibrates, but does not light up.

### Response in Preference Pane
Within the preference pane it says "No device found"

### Detailed Explaination of Issue
I have an xbox one controller, with the latest version of mac and the latest version of 360controller - I use a standard xbox one micro usb cable, it is not a charge cable as i am able to use the controller on windows pc. When plugging in the controller the controller vibrates but does not light up - when i press the xbox button on the controller it flashes but does not connect. I am using a MacBook Pro. The controller is visible via About this mac > System Report > USB and is shown as "Controller". It is shown as a verified microsoft product with the ID's above. Despite this it is still not visible via the Preference Pane in 360Controller. I have clicked "Enable Driver" and typed in my password to allow the change. However after exiting "Advanced" in the 360Controller panel it resets and I have to re check it. I am unsure as to whether this is an issue with my controller, an issue with mac OSX or with the 360Controller driver. I don't believe i have any controller driver's enabled as I have not installed them.

I will be happy with any help I can get - Thanks

FranticRain, 2018-08-29:
I need you to open up your terminal and run `sudo kextload /Library/Extensions/360Controller.kext` and tell me what it says.

If it says `/Library/Extensions/360Controller.kext failed to load - (libkern/kext) system policy prevents loading; check the system/kernel logs for errors or try kextutil(8).` Then you'll need to allow the driver in the `Security & Privacy` section of your `System Preferences`.
[image]

If it responds with something else, let me know and I'll look into it.

georgeawhite, 2018-08-30:
Hey, I entered that into the terminal and it came up with the error message that you have shown but within the Security and Privacy tab there is no option to allow:

[image]
[image]

FranticRain, 2018-08-31:
The message is only present for a short period after connecting a device that the driver should match to. If you connect your controller, it should make the option re-appear.

georgeawhite, 2018-08-31:
That doesn't seem to be working, I'm going to try and uninstall the driver and then reinstall to see if it refreshes it.

theLuthier, 2018-08-31:
I believe I'm experiencing the same/similar issue. Though I see a generic security warning but nothing seems to happen whenever I click "Allow".

[image]

[image]

[image]

[image]

georgeawhite, 2018-08-31:
It may just be an issue on apples behalf, I am tempted to do a system reset but I holding back on doing so. I'm hoping FranticRain may be able to pull together some kind of solution.

All help is appreciated.

FranticRain, 2018-09-01:
So, we have two separate issues here.

@theLuthier You are using some sort of program/input device that Apple doesn't like. It might be a Wacom tablet or a program like MagicKeys. Disable or unplug these devices/programs and try again. That should resolve your problem.

@georgeawhite I got complacent when you told me that it was the same error and didn't look at your message close enough. It is, in fact, not the same error that I told you about. The error you have is a currently unsolved issue related to ownership of your file system. I'm not entirely sure how to fix it. Everyone who has it helps out and tries to debug for a while and then ultimately gives up. I have been completely unable to reproduce it. Basically, the error is telling you that the system thinks that you do not have ownership of the files on your system. I don't know how people manage to remove the kernel from root:wheel, but that is what has happened. I have no idea how to fix it.

georgeawhite, 2018-09-02:
Apologies for the misunderstanding, I will be more than happy to work with you to find a solution, I’ll be the test dummy just tell me what to do

FranticRain, 2018-09-02:
The main problem is that I'm still not entirely sure how to fix it. I'd be curious if the re-format fixes it, but I'm honestly not entirely sure how this happens.

theLuthier, 2018-09-04:
@FranticRain Ahh I do have a wacom tablet, a webcam, and a mic plugged in as well. I'll remove these and try the controller again. Thanks for the advice!

StephenBrough, 2019-09-03:
I just had this issue and it seemingly fixed itself when I ran an update to 10.13.6. I'm on a work-provided Macbook Pro, so I thought IT had done something with permissions, causing the issue. 

First restart after the update finished, I immediately got the security pop up about Drew Mills. Once I allowed that, the app started recognizing my controller. Haven't tested it quite yet, but it's definitely looking more promising.