360Controller/360Controller #737: More of a request rather than an issue. Relevant information provided

unun117, 2018-08-30:
### Type of Controller
Xbox 360 Guitar Hero Live guitar with 360 dongle

### OS Version
10.13.6

### Driver Version
0.16.8

### Connection Method
Bluetooth USB dongle made for Guitar Hero Live on Xbox 360.

### Device Name and Info
N/A

### Response of Controller
The LED on the dongle doesn't flash. It flashes, then it remains lit after the sync button is hit on the guitar when on Windows 10 and Linux.

### Response in Preference Pane
Because it doesn't sync to the dongle, nothing works. 

### Detailed Explaination of Issue
The only driver that works for Guitar Hero Live guitars on PC/Mac/Linux is on Windows 10 using a GHL Xbox 360 dongle specifically. This is to play Clone Hero Live, which is a Guitar Hero tribute program using the Unity engine. The dongle is different than the 360 USB controller adapter. If I can get over the hurdle of activating the USB dongle, I may be able to map the guitar buttons using the existing work you have done for the regular controllers. There is a way to map the controller in the game as well, so I am hoping that if the mapping is the same, the guitar will work. I understand that your existing driver wasn't developed to work with GHL guitars, but it would be greatly appreciated if it could be updated to work for them as well. I am also willing to help in any way I can, because I don't expect you to have the guitar and dongle to test things. If you can't help, I understand. Thanks in advance.

FranticRain, 2018-08-30:
For this issue, "Device Name and Info" is definitely not N/A. I'm going to need the VID and PID of that dongle. Please follow this process and retrieve this information for me. https://github.com/360Controller/360Controller#how-do-i-find-my-vendor-id-and-product-id

unun117, 2018-08-30:
I'm sorry for not being home when I posted the issue. 

The PID is 0x070b
The VID is 0x1430  (RedOctane)

FranticRain, 2018-08-31:
Give this build of the driver a shot. I'm not entirely sure if it will work or not, but it's the best option I can provide for this. Any further work will have to be done by someone else, because I don't support the wireless Xbox 360 code.

[Install360Controller.pkg.zip](https://github.com/360Controller/360Controller/files/2338467/Install360Controller.pkg.zip)

unun117, 2018-08-31:
There was some progress, but it didn't work unfortunately. The dongle flashes, but doesn't sync to the guitar. Weirdly, I got the my PS4 dongle to sync with the guitar, but there is no button functionality. My XB1 dongle doesn't do anything. I really appreciate how quickly you replied and came up with a possible solution. I will keep trying other things.

closed 2018-08-31 by unun117