360Controller/360Controller #739: Request to add controller

jhoff, 2018-08-31:
### Type of Controller
Xbox 360 Midi Pro-Adapter

### OS Version
High Sierra

### Driver Version
0.16.8

### Connection Method
Wired

### Device Name and Info
Device Name: Harmonix Drum Kit for Xbox 360:
Product ID: 0x1138 (4408)
Vendor ID: 0x1bad (7085)

### Response of Controller
Green ring lights up for a second then turns off

### Response in Preference Pane
No Devices Found

### Detailed Explaination of Issue
Reference pane doesn't find device but System Report does. Other controllers work just fine.

Here is the full system report:

Harmonix Drum Kit for Xbox 360:

  Product ID:	0x1138
  Vendor ID:	0x1bad
  Version:	0.28
  Serial Number:	17d7f64a
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Mad Catz Inc.
  Location ID:	0x14200000 / 7
  Current Available (mA):	500
  Extra Operating Current (mA):	0

jhoff, 2018-08-31:
Actually, while we're at it, the device has a switch toggles the type of midi input between drums, keyboard and guitar, so these should be added as well:

Harmonix RB3 MIDI Keyboard Interface for Xbox360:

  Product ID:	0x1338 (4920)
  Vendor ID:	0x1bad (7085)
  Version:	0.28
  Serial Number:	17d7f64a
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Mad Catz Inc.
  Location ID:	0x14200000 / 11
  Current Available (mA):	500
  Extra Operating Current (mA):	0

Harmonix RB3 MIDI Guitar Interface for Xbox360:

  Product ID:	0x1538 (5432)
  Vendor ID:	0x1bad (7085)
  Version:	0.28
  Serial Number:	17d7f64a
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Mad Catz Inc.
  Location ID:	0x14200000 / 12
  Current Available (mA):	500
  Extra Operating Current (mA):	0

FranticRain, 2018-09-01:
Here is a release of the driver that contains the devices requested. I do no know if they will work by default, because they likely use a non-standard packet. If they don't we will need to do some more complicated work to get them working.

[Install360Controller.pkg.zip](https://github.com/360Controller/360Controller/files/2342557/Install360Controller.pkg.zip)

jhoff, 2018-09-04:
Awesome! Thank you! I was able to verify that the unit works like a controller and it seems to be seeing the midi input as well.

closed 2018-09-04 by jhoff