360Controller/360Controller #740: XBOX One USB wired controller not being recognized.

daynerr89, 2018-09-01:
### Type of Controller
Xbox one controller, connected with USB, model# 1537

### OS Version
10.13.6

### Driver Version
360Controller 0.16.8

### Connection Method
Wired. Also tried bluetooth.

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID:
<br/>Vendor ID:

### Response of Controller
searches for pairing but doesn't pair or recognize. It will receive power from the machine but no connection.

### Response in Preference Pane
the controller does not appear in the preference pain.

### Detailed Explaination of Issue
I followed all of the correct steps. Installed > restarted > ensured that I "allowed," in privacy settings. Connected via USB, click "enable" in the controller app. The controller will receive power and search, but will not connect. dropdown says "no devices found". 

Please help, thanks for your support and all you do!

FranticRain, 2018-09-01:
Okay, a couple things. This driver is not involved in the Bluetooth pairing process at all. The preference pane will always report "No device found" for Bluetooth connected controller. They are outside the scope of the driver and work within the operating system without the support of the driver. When connecting the controller via Bluetooth, navigate to a gamepad tester in your browser (http://html5gamepad.com/). You will be able to test your controller and see that it works with the operating system. Game by game support is the responsibility of the game developers and is not our responsibility.

For the wired method, you will need the driver and in order to further assist you, I will need your vendor and product IDs that you omitted from the original issue. Those can be found following these instructions: https://github.com/360Controller/360Controller#how-do-i-find-my-vendor-id-and-product-id

FranticRain, 2018-10-21:
@EXPOSEDKING As I always say to people, please create your own issue instead of copying other people's issues. Most people have individual issues that are not solved by piggybacking on existing issues. I will respond to your issues on an actually relevant post for your actual controller instead of this wholly unrelated issue.

This issue is being closed due to inactivity.

closed 2018-10-21 by FranticRain