360Controller/360Controller #741: No Devices found in Xbox 360 tab in System preferences

PhillipRo, 2018-09-01:
Original , Xbox 360

### OS Version
High Sierra - 10.13.5 / Macbook Pro 2012

### Driver Version
360Controller 0.16.8

### Connection Method
Wired

### Device Name and Info

Product ID:	0x028f
Vendor ID:	0x045e  (Microsoft Corporation)
Version:	1.05
Speed:	12 Mb/seg
Location ID:	0x14200000 / 16
Current Available (mA):	500
Current Required (mA):	100
Extra Operating Current (mA):0

### Response of Controller
Lights up

### Response in Preference Pane
No Devices found in Xbox 360 tab in System preferences

### Detailed Explaination of Issue
No Devices found in Xbox 360 tab in System preferences

FranticRain, 2018-09-01:
Please ensure that the driver has been allowed in System Preferences -> Security and Privacy -> General. This approval UI is only present in the Security & Privacy preferences pane for 30 minutes after the alert. Until you approve the kext, future load attempts will cause the approval UI to reappear but will not trigger another user alert.

[image]

PhillipRo, 2018-09-01:
I already did this, but it still does not work

FranticRain, 2018-09-01:
I need you to open up your terminal and run `sudo kextload /Library/Extensions/360Controller.kext` and tell me what it says.

PhillipRo, 2018-09-01:
He asked for the password, I typed and then appeared: Computername:˜user$

mmun, 2018-09-02:
I'm in a similar situation and I get

```
/Library/Extensions/360Controller.kext failed to load - (libkern/kext) system policy prevents loading; check the system/kernel logs for errors or try kextutil(8).
```

BTW, clicking "Allow" in the screenshot above doesn't seem to do anything.

FranticRain, 2018-09-02:
@PhillipRo So that means that the driver is being correctly loaded. If you could please try using the controller after running that command, could you tell me if the driver still isn't working for you? All signs point to the driver working fine on your system, and your device is already included in the supported devices, so I'm not sure what the problem is here.

@mmun As stated in the README, "This prompt has been known to have issues with software or hardware that interferes with mouse movement. If you are using software that impacts the movement of your mouse, such as MagicKeys, or are using a special interface device, such as a Wacom tablet, please using a standard input device, such as a mouse, to click this button. This is a security feature of macOS and is out of our control."

alexandre-thauvin, 2018-10-17:
Hello @FranticRain 
I'm in a similar situation and still no devices found after running kextload (btw no error on this command).

[image]

FranticRain, 2018-10-18:
@alexandre-thauvin Please make a new issue and include your controller product and vendor information.

iikas, 2018-12-26:
Where can I get the version 360Controller 0.16.8? HELP!! 
already search for many places and nothing

FranticRain, 2018-12-26:
@iikas Please consult the README for the answer to your query.

FranticRain, 2019-04-06:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-04-06 by FranticRain