360Controller/360Controller #742: xbox one controller, not finding any devices

DxTillett, 2018-09-01:
### Type of Controller
Replace this text with: Original Xbox, Xbox 360, Xbox One, All or N/A

### OS Version
Replace this text with the version of macOS that you are running

### Driver Version
Replate this text with the version or versions of the driver that you are using or have tested

### Connection Method
Replace this text with: Wired, Wireless Adapter, Bluetooth, N/A or more specific method if applicable

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID:
<br/>Vendor ID:

### Response of Controller
Replace this text with information on how the physical controller reacts. Does the guide button light up? Does it vibrate? Etc.

### Response in Preference Pane
Replace this text with information about the controller's reaction in the preference pane. Does the controller appear in the preference pane? Do all of the buttons, joysticks, and triggers act as expected?

### Detailed Explaination of Issue
Replace this text with a detailed explaination of your issue. Simply saying "it doesn't work" is a quick way to ensure that your issue is never fixed. Without detailed information on your specific issue, we cannot fix your issue. Including as much information as possible in this section increases the likelyhood of getting your issue quickly and effectively resolved.

FranticRain, 2018-09-09:
Is this the new adaptive controller? Or some other type of controller? If it's an adaptive, then this may take some more work. Otherwise, I would just like to know why there is a new Xbox One controller, just in case it causes some fresh new issues. Here's a build with the controller added.

[Install360Controller.pkg.zip](https://github.com/360Controller/360Controller/files/2363928/Install360Controller.pkg.zip)

d235j, 2018-09-09:
Doesn't sound like the adaptive controller (which lacks vibration features)

FranticRain, 2018-09-09:
@jhkrischel Xbox One controllers are supported via Bluetooth through the operating system. Bluetooth connected Xbox One controllers will not show in the preference pane. They are outside the purview of this driver. If a game doesn't support your controller in Bluetooth mode, it is the responsibility of the game developer to properly support controllers, not ours. Furthermore, wireless is not supported for Xbox One controllers. No one has cracked the obfuscation scheme that Microsoft is using with the wireless adapter.

Looks like this works just fine, still waiting for an adaptive controller to roll around because that is almost certainly going to have a new packet.

closed 2018-09-09 by FranticRain

FranticRain, 2018-09-14:
@the-t-in-rtf I'll need the product and vendor IDs as per usual, instructions are in the README. In addition, I'm going to need packet sniffs, as documented here: https://vestigl.com/sniffing/index.html. Except you'll need to document what order you press the buttons and such in. And if you have a device that works with the 3.5mm jacks, getting readings from the furthest left and furthest right inputs should get me everything I need. I don't know how in the world we are going to make it work, but we'll see.

It may require a whole new remapping interface, which I'm not too stoked about. But I want to do my due diligence on making this thing work.

FranticRain, 2018-09-15:
@the-t-in-rtf Yes, please make a new issue.

the-t-in-rtf, 2018-09-17:
@FranticRain, [here's the new issue](https://github.com/360Controller/360Controller/issues/753).