360Controller/360Controller #743: Xbox 360 afterglow controller, no devices found

HFrostly, 2018-09-02:
### Type of Controller
xbox 360 wired afterglow controller 
### OS Version
Mac OS high Sierra version 10.13.6

### Driver Version
Xbox 360 Controller Driver Version 0.16.8

### Connection Method
wired, into a female usb 3.0 port that goes into a male usb c, that goes into the MacBook Pro.

### Device Name and Info

  Product ID:	0x0213
  Vendor ID:	0x0e6f
  Version:	1.00
  Serial Number:	069B0B3F
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Performance Designed Products
  Location ID:	0x14100000 / 1
  Current Available (mA):	500
  Extra Operating Current (mA):	0

### Response of Controller
when plugged in, the controller lights up for a second. the middle xbox Button blinks, then nothing. 
### Response in Preference Pane
says "no devices found"

### Detailed Explaination of Issue
I bought a connector and tried hooking it up with the wired afterglow controller, it doesn't work. I have tested my phone through the connector and it transmits data through it.  
for anyone wondering the connector specs on the box say "rated for 15w charging" and "usb 3.0 5Gbps transfer speed" (idk if this helps at all). 
Im trying to play games like rocket league on my mac. 
I have also allowed the use from "Drew Mills" developer software

HFrostly, 2018-09-06:
alright after re-instalation the "allow" button doesn't pop up. anyone know how to fix this?

FranticRain, 2019-01-06:
Please ensure that the driver has been allowed in System Preferences -> Security and Privacy -> General. This approval UI is only present in the Security & Privacy preferences pane for 30 minutes after the alert. Until you approve the kext, future load attempts will cause the approval UI to reappear but will not trigger another user alert.

[image]

avargas715, 2019-01-24:
I'm having the same issue and I already allowed the driver.

FranticRain, 2019-01-24:
@avargas715 Make a new issue and include the result of running `sudo kextutil /Library/Extensions/360Controller.kext` in your issue.

FranticRain, 2019-05-26:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-05-26 by FranticRain