360Controller/360Controller #744: Will Thrustmaster T-Flight Hotas X Flight Stick work with this driver?

Spudboy, 2018-09-02:
Sorry if I’m doing this wrong.

I just wanted to know if 360Controller will work with a Thrustmaster T-Flight Hotas X Flight Stick?
I’m thinking about buy this to play Overload on my iMac.

Thanks for any help.

FranticRain, 2018-09-02:
If the controller doesn't work with an Xbox branded console, then it won't work with this driver. So in your case, it will not work. Sorry.

Spudboy, 2018-09-02:
The controller works with Xbox. That is why I figured it would work in conjunction with 360Controller.

FranticRain, 2018-09-04:
Sorry, in my search for this controller, I didn't find an Xbox compatible one, but on second search I found it. I need you to follow the issue template in order to add this controller. It's a fairly standard device request and requires the same template. Copy the content of this link and edit it to match your requirements. Paste it as a comment in this thread. https://raw.githubusercontent.com/360Controller/360Controller/master/issue_template.md

FranticRain, 2018-09-10:
Yes. This driver fully supports Xbox One controllers. It may require additional work on your end, because often these types of devices have slightly different packet formats than a standard controller. For example, fight sticks and racing wheels both have non-standard packet layouts on Xbox One.

Spudboy, 2018-09-11:
Now I’m unsure if I should get the Thrustmaster VG Thrustmaster T-Flight HOTAS One since it is not supported by Rewired™ and the Thrustmaster T.Flight Hotas 4 Flight Stick for PS4 & PC is already supported and works with Overload (the game that I’m wanting to use with a flight stick). I don’t want to waste your time. I need to do more research.

closed 2018-09-12 by FranticRain