360Controller/360Controller #745: Driver version 0.16.8 has massive bugs

L0VEFIRE, 2018-09-06:
### Type of Controller
Xbox One

### OS Version
macOS High Sierra / macOS Mojave

### Driver Version
0.16.8

### Connection Method
Wired - (Micro USB to to USB cable) plugged into (Apple® USB to USB-C Adapter)

### Device Name and Info

**Controller: (Xbox One)**

  **Product ID:**	0x02ea
  **Vendor ID:**	0x045e  (Microsoft Corporation)
  Version:	3.01
  Serial Number:	3032363030333738393633363434
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Microsoft
  Location ID:	0x14100000 / 7
  Current Available (mA):	500
  Current Required (mA):	500
  Extra Operating Current (mA):	0

**iBridge: (Apple USB to USB-C Adapter)**

  Product ID:	0x8600
  Vendor ID:	0x05ac (Apple Inc.)
  Version:	1.01
  Manufacturer:	Apple Inc.
  Location ID:	0x14200000

### Response of Controller
Physical issues, have to due with vibration; if I have vibration on both or triggers, the controller will vibrate, and the left side will just stop then not work anymore, eventually same happens to right. then it  will come back randomly if i wait long enough or if i replug the controller in.

### Response in Preference Pane
N/A

### Detailed Explaination of Issue
When playing any game that requires this driver, I have noticed a couple odd issues:
1) in fortnite i had an issue where i couldnt move to the left or right or forward or back, its like as if deadzones where on, but there not. that last for 20-30 seconds and then it went away.
2) the controller will intermittently out of the blue stop responding to actions, then resume happens at least 2-3 times each time i use the controller.
3) when playing with the controller a more frequent issue is i be moving my player forward find a place to hide, then sudden the screen starts to slowly pan to the right or left, like the controller joystick is stuck but its fine. this also last around 10-30 seconds (varies)

i had less issues in 0.16.6 then 0.16.8 (the ghost panning still happened here, but the random deadzone issue never happened)

** If you need logs or any supporting files to help let me know.

FranticRain, 2018-09-06:
Okay. There's a whole lot to unwrap here.

Rumble stuff: That's just how the Xbox One controllers respond to rumble. If you rumble them for too long, like can happen in the preference pane, the controller will assume that something has locked up and will disable rumble. The rumble re-activates because you've caused an internal counter to overflow eventually, resulting in a return to service. That's just a controller feature, and not something I'm interested in engineering a workaround for.

As for your input issues, I dug through all of the commits since 0.16.6 in order to try and find where these bugs might have been introduced. There is only one change in the input path, and it should only be impacting input when you press the guide button on your controller.

Now. What is really causing these issues?

Most likely:
1. Fortnite
2. USB-C adapter. Not much we can do here. macOS is just super finicky with drivers and USB-C adapters. It likes to cut out every once in a while, just to keep you guessing.
3. Controller drift caused by inadequately sized deadzones. Increase the size of the deadzones within the driver.

Less likely:
1. I assume other button inputs work during this period. Based on the changes since 0.16.6, you would have to be experiencing some sort of temporary settings corruption that results in a deadzoneLeft and deadzoneRight both being simultaneously corrupted and being set to their maximum value. This case is nigh on impossible.
2. The `reserved1` variable of the `XBOX_IN_REPORT` is actually the MSB of the `counter` variable. That would result in a situation where the the MSB of counter finds its way to 0x30 and all of the button inputs are read as guide button reports. The real question here is if this occurs in all games, or just Fortnite specifically. If you could reproduce this issue in the preference pane, I'd be inclined to believe that this is the cause. Another way to test it would be to spin both sticks and pull the triggers as much as you can during the outage. This should result in a faster return-to-service than pressing buttons only. Which would result in an even faster return-to-service than not pressing anything at all.
3. Controller fails to send a final report to the driver that puts your input back into the deadzone range. This would mean that the drift is always in the direction that you returned your controller from. So if you looked left, and released your stick, the drift would be left. This would mean that there is no period in which the view is static and then begins to drift.

Were your experiences with 0.16.6 and 0.16.8 both while playing Fortnite? Or were you playing different games at the time? Have you experienced any of these issues outside of Fortnite?

FranticRain, 2018-09-12:
With the temperature thing, it's highly likely that the game is hogging resources, and not giving the driver the resources it needs to process inputs. It's possible that if you had some fancy mouse that required a driver (Like a Logitech MX Master, for example) you would actually experience drifting with it as well. Basically, the driver doesn't have a chance to provide the packet that says "Hey the stick is back to zero" in time, so it is dropped, and just ignored. Resulting in your issue. I wouldn't be surprised to also see the driver dropping button and trigger inputs, though the stick movement is more likely since it floods more packets to the driver.

As for the video cut out thing, that's almost 100% a 10.11 and later bug. The sort of thing that would only get fixed with a re-write. I'm pretty sure we won't be able to help you with that.

closed 2018-09-14 by FranticRain