360Controller/360Controller #746: My computer shuts down when I disconnect my receiver or power off my controller

Will-Do-Like-Mildew, 2018-09-07:
### Type of Controller
Xbox 360

### OS Version
High Sierra 10.13.6

### Driver Version
0.16.8

### Connection Method
Wireless Adapter

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Donop Wireless USB Gaming Receiver for Xbox 360 Controller to PC
<br/>Product ID: Doesn't show up
<br/>Vendor ID: Doesn't show up

### Response of Controller
Controller works absolutely fine.

### Response in Preference Pane
Nothing is wrong here.

### Detailed Explaination of Issue
Recently updated to High Sierra from El Capitan. Ever since the update, clicking the power button in the Xbox 360 Controllers menu shuts down my computer. When a controller is connected, disconnecting the wireless receiver from my computer also shuts down my computer. This did not happen prior to updating my OS. If it helps, my wireless receiver was ordered from here: [https://www.amazon.com/wireless-Gaming-Receiverfor-Xbox-360-Controller/dp/B00BV72X1U](url)

macmeister1967, 2018-10-21:
I have a very similar issue, except it is version 10.16.9 running on 10.13.6 and the computer **RESTARTS** instead of shutting down. I have had other issues with my Logitech Mouse cutting out, or cursor wander, and I think it is due to MICE/360Daemon.app :-(

FranticRain, 2018-10-21:
As of macOS 10.11, Wireless controllers have been unsupported by this driver. This is thanks to a rewrite of the macOS USB kernel. This issue can only be resolved through a rewrite of the driver, which is currently unplanned. Therefore I will be closing this issue, since it is outside the realm of supported devices.

closed 2018-10-21 by FranticRain