360Controller/360Controller #748: Driver does not recognize 3rd-party wired Xbox 360 controller

drpre, 2018-09-13:
### Type of Controller
Xbox 360

### OS Version
High Sierra 10.13.6

### Driver Version
Tried 0.16.8 (latest signed release) and 0.16.3 (what I had installed before)

### Connection Method
Wired

### Device Name and Info
>Device Name: Metallics LE Xbox 360 Controller
<br/>Product ID: 0x0159 (decimal: 345)
<br/>Vendor ID: 0x0e6f (decimal: 3695)

### Response of Controller
When very first plugged in the entire ring of the guide button lights up for half a second, then the lights go out and the controller does nothing.

### Response in Preference Pane
Controller does not appear in "Xbox 360 Controllers" page -- the drop-down says "No devices found"

### Detailed Explanation of Issue
I have used a 1st-party wired 360 controller with this laptop and driver and it works perfectly. However I just replaced with a 3rd-party controller from Performance Designed Products and the controller does not show up in the "Xbox 360 Controllers" preferences pane, and does not respond to any button presses.

I have tested other 1st-party controllers with my Macbook and they continue to work perfectly. Conversely, I have tested this 3rd-party controller on a Windows machine and it also works.

I am fairly certain that the Info.plist file just needs to be updated with the vendorID/productID tuple for my controller -- I see several other entries with the same vendor ID, but none with a matching product ID.

===

Based on other similar threads I will tell you this proactively: there is no output from running `sudo kextload /Library/Extensions/360Controller.kext` -- I imagine this proves the driver is loading without problems.

==

Happy to provide any additional info as requested!

FranticRain, 2018-09-14:
Here's a build with the controller added. Sorry for the .pkg.zip. Github doesn't like pkgs.

[Install360Controller.pkg.zip](https://github.com/360Controller/360Controller/files/2381576/Install360Controller.pkg.zip)

closed 2018-09-14 by FranticRain

drpre, 2018-09-19:
Thank you very much for delivering the fix so quickly!