360Controller/360Controller #749: Hi, unable to connect xbox one controller wired to MacBook Pro, is this possibly because I am using a USB-C adapter?

tobyb434, 2018-09-14:
### Type of Controller
 Xbox One

### OS Version
MacOS High Sierra version 10.13.6

### Driver Version
0.16.8

### Connection Method
Wired with USB C to USB A adapter 

### Device Name and Info
Macbook Pro (2018)

### Response of Controller
Doesn't provide response to connection to Mac

### Response in Preference Pane
Not Connecting 
### Detailed Explaination of Issue
Trying to connect xbox one controller to MacOS, using a USB C multiport adapter to USB A plugged into Micro USB xbox one controller. System is not recognising the xbox one controller connection at all, is this because I am using a multiport USB C adapter?? Thanks

FranticRain, 2018-09-15:
It's possible that it is the USB-C adapter. They are known to cause issues with this driver. It is also possible that you are using a USB-A to Micro USB cable type known as a "charge" cable. These cables do not have wires connecting the data lines of USB, and instead only provide power to a device. Without these data lines, the computer cannot talk to the controller, and thus the driver can't do anything.

At this time, I am not aware of any particular USB-C adapters that work with the driver, though I am confident that some exist.

closed 2018-09-15 by FranticRain

schumannd, 2021-01-10:
Are there any news on this issue? It would mean that the it's currently not possible or unknown how to use the 360controller with any of the new macbooks