360Controller/360Controller #750: worked at first, not anymore

richardfreemanlol, 2018-09-14:
Xbox One wired
High Sierra 10.13.16
latest version of driver

worked for about a day, then I restarted my computer and it no longer worked. it says no devices found, i have enable driver checked, but act as an xbox controller is greyed out. in security and privacy> general, it doesn't say that I have to allow the driver, so I am just confused on every issue lol

FranticRain, 2018-09-15:
If you've allowed the driver once, then you shouldn't have to do it again. I'll need to have you complete some debugging steps so we can find out what has gone wrong. [First, follow the instructions in the README to find your vendor and product IDs and provide those to me.](https://github.com/360Controller/360Controller#how-do-i-find-my-vendor-id-and-product-id) This will let us know if your controller is being recognized within the OS. If you find your vendor and product IDs, then I need you to run this command in your terminal:
```
sudo kextutil /Library/Extensions/360Controller.kext
```
If it gives you an error message, please post it here. If it doesn't give an error message, your controller should work properly.

Limpybanana, 2018-10-01:
Okay, I have the exact same problem that he did. I did the steps that you said, and this is what I got...

Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

Limpybanana, 2018-10-01:
Also, steam recognizes it, and the xbox one driver (not 360) recognizes that there's a controller connected. In system report > USB, the 'Controller' is even there. Product ID: 0x02d1, Vendor ID: 0x045e  (Microsoft Corporation). I've done what I think most people have already suggested to be the source of the issue. I have yet to have asked for help, but now I'm stuck and don't know what to do. It worked fine until I opened up Steam's Big Picture Mode (BPM) and it asked me to update it (I think it asked me to update the controller, possibly), It then told me to reset my computer, so I did, and now the Xbox 360 Controllers in system preferences won't register that there's a controller connected.

BUMPRW, 2018-10-03:
> When you say that the "Xbox One Driver" recognized that there's a controller connected, what does this mean? But this, do you mean that Xone-OSX picks it up? Because you DEFINITELY shouldn't have that installed.

OS 10.13.6 picks up the device, but you can’t do anything with it. The lights do come on, but immediately turn off.  I did not have the driver installed.  I then installed the driver and still nothing happened, so I tested your driver with my Saitek AV8r-02 and it worked fine.  I did attempt to follow your build instructions, but without a signing certificate XCODE wouldn’t let me do anything.  Looks like Apple is ending our days of experimenting and having fun playing with the software.

FranticRain, 2019-01-15:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-01-15 by FranticRain