360Controller/360Controller #751: Cannot uninstall driver / reinstall driver / use driver, and system no longer shuts down/restarts properly

justplainmark, 2018-09-16:
### Type of Controller
Wired XBOX 360 controller (circa 2007ish I think?)

### OS Version
10.13.4 (17E199)

### Driver Version
Most recent 0.16.9

### Connection Method
Wired 360 controller 

### Device Name and Info
I referred to the ReadMe file and was not able to find the Device name, Product ID, or Vendor ID. My best guess is that the wired controller I have has some short in the wire or connection issue, as it's not able to consistently remain connected to my Mac for more than a few seconds (and even then it doesn't show up in the USB tree). This is secondary to the main issue (see "Detailed Explanation of Issue" below)
>Device Name: Not listed
<br/>Product ID: Not listed
<br/>Vendor ID: Not listed

### Response of Controller
The controller lights up briefly when initially plugged in, or when I apply pressure/jiggle the secondary connector in the cord. I'm guessing the controller is bad. This is secondary to the main issue (see below) - at this point I'm not really trying to get the controller to work as I am trying to uninstall the driver and fix my computer.

### Response in Preference Pane
Sometimes the controller shows up in the reference pane of the System Preferences > XBOX 360 Controllers, but rarely for more than a second or two before disappearing. 

### Detailed Explaination of Issue
<b><u>Cliff Notes Version</u></b>: A few days ago, my controller started having weird lags in responsiveness that made gaming difficult (in retrospect, this was likely failing hardware as the controller is 10+ years old). I reinstalled the XBOX 360 drivers two days ago and am now unable to uninstall the driver, reinstall the driver, or shut down my mac properly. At this point I'm more interested in removing the driver from my machine than trying to get it to work.

<b><u>Full version</u></b>: I attempted to reinstall/update the XBOX 360 driver to fix laggy responsiveness during gameplay (<i>Enter the Gungeon</i> is hard enough to begin with and becomes impossible with delayed controls). When the initial reinstall did not solve the problem, I uninstalled the drivers entirely, restarted the machine, and then attempted to reinstall them. 

On reinstallation, the installer hung up and froze near the end of the process just before installation completion with "Install time remaining: Less than a minute" under the process bar. I left it for an hour or two and it was still hanging there, so I attempted to force quit the application with CMD + ALT + ESC. This did not work, so I tried to restart the machine. The restart process halted with just the mouse cursor visible against a black background, with no interactivity or function available. I manually force shut down of the computer by holding the power button. 

On restart, the XBOX 360 driver options are available in the System Preferences window but I'm unable to uninstall them. I get a "Unable to Uninstall the driver!" error (see screenshot). 

[image]

Further, attempting to reinstall the drivers causes the Installer to halt at exactly the same moment that it did previously, and no amount of time will allow it to finish (I've tested it for up to 24 hours).

[image]

The XBOX 360 wired controller will no longer connect for more than a second or two. Restarting the machine causes it to halt on a black screen with just the mouse cursor visible. Only a force shut down by holding the power button can remove it from this state (put bluntly, I can no longer restart normally and this is very concerning).

Any advice?

Thanks,
-Mark

FranticRain, 2018-09-16:
Jump in to your terminal and run these commands:
```
sudo launchctl unload /Library/LaunchDaemons/com.mice.360Daemon.plist
sudo kextunload -b "com.mice.driver.Xbox360Controller"
sudo kextunload -b "com.mice.driver.Wireless360Controller"
sudo kextunload -b "com.mice.driver.WirelessGamingReceiver"
sudo rm -f  /Library/LaunchDaemons/com.mice.360Daemon.plist
sudo rm -rf "/Library/Application Support/MICE/360Daemon.app"
sudo rm -rf /System/Library/Extensions/360Controller.kext
sudo rm -rf /System/Library/Extensions/Wireless360Controller.kext
sudo rm -rf /System/Library/Extensions/WirelessGamingReceiver.kext
sudo rm -rf /Library/Extensions/360Controller.kext
sudo rm -rf /Library/Extensions/Wireless360Controller.kext
sudo rm -rf /Library/Extensions/WirelessGamingReceiver.kext
sudo rm -rf /Library/Extensions/XboxOneBluetooth.kext
sudo rm -rf /Library/PreferencePanes/Pref360Control.prefPane
sudo pkgutil --forget com.mice.pkg.Xbox360controller
```
This is the manual equivalent of the uninstall button. Then, run:
`ls /Library/Extensions`
In the list it returns, you shouldn't see anything related to the driver. That should get you back to a good state, and then we can work on the rest from there. Let me know how it works out.

justplainmark, 2018-09-18:
Okay so the "about two minutes" update message didn't go away after another 15 minutes or so, so I force shut down the Mac by holding the power button. The machine started up just fine, though slightly longer on the boot up than normal (maybe 30 seconds longer? I wasn't timing it). 

The 360 Controller option is gone from the System Preferences panel, but the machine cannot restart normally without locking up on the black screen. This problem is consistent and happens every time I attempt to restart. On locking up on the black screen, the mouse cursor is still visible and can move around but no action is possible. The only way to leave this state is to force restart.

This is kind of a reach, but I'm going to guess that the system is trying to update something on restart but is hanging because the files aren't there? Or something? I dunno. I used to do PC repair like a decade ago but this is escaping my understanding.

Any suggestions? I have AppleCare so worst comes to worst I can just make a Genuis Bar appointment.

FranticRain, 2018-09-18:
You've successfully removed all of the driver components, and unloaded it all by rebooting. The driver no longer on your system. There's nothing more we can do for you at this point, regarding your boot situation. It doesn't appear that the driver is at fault.

LightSystem, 2018-11-04:
I've had the same issue with freezing on shutdown/restart just as described here and have found a way to fix it.
I have tried many things I'm not sure which one worked but after trying to uninstall and/or running the uninstall commands above, follow that with:
- Search for any unwanted kernel extensions that might be still be around. Probably the easiest way is to just search the whole system `sudo find / -name "*.kext"` and the text search the result list for keywords like `360` or `xbox`.
- Delete all the extensions you can find related to the controller driver obviously being careful not to delete any kexts that you actually need.
- You might see some `.kext` you cannot delete, that's because they are staged by the OS, leave these for now, and delete all the other ones.
- When finished run `sudo kextcache -invalidate /`, this will get rid of the staging of the extensions you've deleted.
- Just restart, your problem should be fixed.

Hopefully this is helpful to someone.

justplainmark, 2018-11-12:
Hey @LightSystem, thank you for replying to this! And also @FranticRain, heads up:

I tried your method and ran into some issues in deleting the files. First, I ran the search protocol you mentioned `sudo find / -name "*.kext"` and found that there are still two .kext files on my computer from the 360 controller install:

```
/Library/StagedExtensions/System/Library/Extensions/360Controller.kext
/Library/StagedExtensions/Library/StagedExtensions/System/Library/Extensions/360Controller.kext
```

I know these look like the same file but the path and folder names are different. Anyway, I'm not able to delete these files. If I try from finder, I get this error:

[image]

And if I try to delete them from Terminal, it tells me `Operation not permitted`

Things I've tried that also didn't work and gave this error:
- `rm [pathname]`
- `rm -rf [pathname]`
- `sudo rm [pathname]`
- `sudo rm -rf [pathname]`
- `chflags noschg [pathname] followed by every variation of removal above`

Other things I've tried
- One forum post suggested that the reason I can't delete it is due to some security feature in OSX 10.11 and newer which doesn't let users remove these. I tried using `csrutil disable` to turn this off and it did not work
- I tried unlocking the files and their parents folders in Finder, but the Lock/Unlock dialog was greyed out and unselectable in the "Get Info" menu which makes me think it's not an option
- I tried restarting in Recovery Mode and removing the files via Terminal there, but the terminal window didn't acknowledge the path names (but yet could still find them with the `sudo find` command).

This is weird. How do I delete these files? I'm out of ideas.

FranticRain, 2018-11-12:
The reason the kexts are in `StagedExtensions` and that I didn't tell you delete them from there, is that they should be removed from that directory on a reboot. That directory is just where macOS stores extensions that you haven't approved yet. So you didn't accept the developer signing certificate during installation. Removing these kexts is not done through `rm` since they're staged. All you have to do is invalidate the kext cache. Can be done with this command:
```
sudo kextcache -invalidate /
```
Those files will now be gone.

justplainmark, 2019-01-15:
So sorry for the delayed response here - @FranticRain your suggestion of the `sudo kextcache -invalidate /` command worked like a charm. The system restarts without issue now. Thank you!

closed 2019-01-15 by FranticRain

DankEnigmo, 2021-07-08:
so I did all of these steps but i want to use my controller on this pc, so i tried to reinstall the drivers and this showed up 
[image]
what can i do to resolve this

prhost, 2021-07-12:
`WirelessOneController.kext` too is extension from 360Controller

jhz27, 2024-05-14:
Omg It’s worked thanks.

EMRestart, 2024-09-01:
Oh my god. I haven't been able to correctly restart my mac in YEARS and had no clue why. Just randomly came across this and it fixed my issue. Absolute lifesaver of a thread. I'm in disbelief.