360Controller/360Controller #753: Add support for Xbox Adaptive Controller

the-t-in-rtf, 2018-09-17:
### Type of Controller

Xbox Adaptive Controller

### OS Version

10.13.6 (High Sierra)

### Driver Version

0.16.8

### Connection Method

Wired

### Device Name and Info

>Device Name: Controller
<br/>Product ID: 0x0b0a
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller

No response (yet).

### Response in Preference Pane

Controller doesn't appear in preferences pane.

### Detailed Explanation of Issue

As discussed in [the request to add support for Xbox One controllers](https://github.com/360Controller/360Controller/issues/742#issuecomment-421525222), I have an Xbox Adaptive Controller and am happy to help troubleshoot/test adding support.

the-t-in-rtf, 2018-09-17:
@FranticRain, I'm forking the source at the moment and adding the new device details.  I don't typically compile things like this (developer, just not a Mac dev), so it may take a bit to get set up.  You asked me to provide data from the leftmost and rightmost buttons, I'll also need a bit to build a button that connects to a 3.5mm jack (already have the parts).  More info as I have it.

the-t-in-rtf, 2018-09-17:
Just to confirm, is Xcode 6.4 still required?

MaddTheSane, 2018-09-17:
Yes, Xcode 6.4 is still needed. We haven't got the rewrite working yet.

FranticRain, 2018-09-17:
You shouldn't need to install Xcode at all. If you just provide me the data, then I can take care of it. No need to learn a new codebase and all that.

But if you insist on doing it yourself, here's some help for getting 6.4 running on modern OS versions. https://github.com/360Controller/360Controller/issues/605#issuecomment-372765092

the-t-in-rtf, 2018-09-18:
@FranticRain, thanks, I appreciate that, as the controller also supports bluetooth and I'd much rather just play with it (it's awesome).  I made a bunch of switches out of arcade buttons last night, so as soon as you have a driver to test, I can give you whatever info you need about the inputs from the jacks.  Let me know if you need any more information than the vendor ID and product ID.

the-t-in-rtf, 2018-09-19:
The only info in system.log is as follows:

> Sep 19 07:30:05 duhrer-macbook System Preferences[838]: ControllerType from prefs was 2, expected 3

FranticRain, 2018-09-19:
As I expected it uses a non-standard packet. Like I said in the previous issue, I need to you sniff the packets of the device on a Windows machine. The instructions are listed here: https://vestigl.com/sniffing/index.html

Without this information, I cannot get this working.

sharkwheels, 2018-09-22:
@FranticRain Hey. Just downloaded that zip. I'm using OSX El Capitan and this update works for me w/ the adaptive controller, so just wanted to chime in and say thank you!

the-t-in-rtf, 2018-09-24:
Here's [a PCAP file](https://www.dropbox.com/s/4awg716onx60baw/2018-09-24.pcap?dl=0) I created in a VM following the instructions you sent.

FranticRain, 2018-09-27:
Okay so here's what I'm seeing, full packet from adaptive:
20 00 07 32 80 00 00 00 00 00 00 00 00 00 00 00 00 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Brackets added to denote each section:
[20 00 07 32] [80 00 00 00 00 00 00 00 00 00 00 00 00 00] [80 00 00 00 00 00 00 00 00 00 00 00 00 00] [00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00]

First four bytes: Xbox Header
[20 00 07 32]
It's got the button packet header, the empty value, the counter and the size, as expected.

Next 14 bytes: Device input?
80 00 00 00 00 00 00 00 00 00 00 00 00 00

Next 14 bytes: Device input?
80 00 00 00 00 00 00 00 00 00 00 00 00 00

Next 22 bytes: Unknown
00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

So based on this packet standard and the code in the driver, this works automatically.

@the-t-in-rtf Just to make sure, right before you ended the capture, you were pressing A B X Y in that order, correct? If so, this driver works automatically and the issue with it not working is specific to something you've done wrong. Can you use other Xbox controllers with the driver and are only unable to get it to work with your adaptive controller?

the-t-in-rtf, 2018-09-27:
The adaptive controller doesn't actually have physical X Y buttons.  I pressed the physical A and B buttons, then switches connected to the X and Y buttons.

The driver does work with existing Xbox 360 controllers I have.  When I connect it as a wired controller, the Adaptive controller shows up in the list, but the preferences pane doesn't give any indication when buttons are pressed, nor does it work in applications.

The controller itself works perfectly when connecting via bluetooth.  When testing that setup, I connect to a power-only port on my hub.  When testing it wired, I remove the paired bluetooth device and connect to a full USB port.

Just to note, the other user reporting this as working wasn't running High Sierra (which I am).  I did have a previous version of the driver installed, I will uninstall, reboot, and reinstall the package you sent just to check that.

the-t-in-rtf, 2018-09-27:
No difference, nor did bypassing the hub change anything.

lwoodbury, 2018-11-26:
Just an FYI, I am having same experience as @the-t-in-rtf on High Sierra. The package above give me the preference pane and with some fiddling the device pops up, however no response to any controls. Via bluetooth all is working.

seamusb, 2019-02-17:
JFYI, I have a tankstick with a trimode PCB in xinput mode (https://shop.xgaming.com/products/pre-order-new-x-arcade-xinput-upgrade-kit) which plugs in via a single USB cable.
It detects as two "Wireless 360 Controller (Wireless)" in version 0.16.9.
No longer detects in 0.16.10. 
I am on mac 10.14.3
Not a problem at all as I am happily using the older version. (Thanks so much BTW)
I figure the adaptive controller changes in 0.16.10 might be causing, so I thought I would mention it here. :) Sorry, this is not super helpful. If I find some time I will look at the diffs.

FranticRain, 2019-02-18:
@seamusb This is a separate problem and you should create a separate issue to discuss it.

FranticRain, 2019-04-01:
@FelixHenninger If the driver sees the controller, then the controller has correctly paired with the OS. None of that stuff really tells me anything. 

By my understanding, this device should "just work" with the driver, based on how the driver is written. Now that it's confirmed not working for multiple people, I will have to develop a version with more debug outputs. I had one user that confirmed that it worked just fine and one that said that it didn't work at all, so I just let it sit for further response. But now that there is another confirmed failure, we can work off of that.

FelixHenninger, 2019-04-02:
Hi @FranticRain, thanks so much for your super-quick response! I'd love to support you in this, would it be helpful if I created another PCAP log from windows? I also have a logic analyzer and could dump raw USB traffic on the mac, if that would be useful to you. Please let me know if and how I can help, I'd be glad to. Thanks again for all of your work on this!

FelixHenninger, 2019-04-02:
Ok, just a quick update: I played around with the adaptive controller some more, and after failing to get any response either on the mac or on linux via `evdev`, I found a Windows PC to connect the controller to, as a last step before sending it back. It took a moment to pair, but after that was complete, it accepted input, and, mysteriously, the controller now works across all systems -- input shows up in your system settings panel, as well as via `evdev`, no problems at all. (maybe, @the-t-in-rtf, this could also work for you?)

I don't have a good explanation for this, so I'll leave it here in case it's of use to someone. I think I previously paired the device via Bluetooth, maybe that threw it off?

In any case, I'm happily enjoying the controller with my mac now. Thanks a lot for making this possible, and sorry for taking your time with my support requests! I owe you some of your preferred beverage, is there any way to contribute a bit to your coffee/beer/tech/... fund, or to help a good cause in your name?

Cheers, -Felix

FranticRain, 2019-04-03:
Best guess is that the adaptive controller took some kind of firmware update when you plugged it in to a Windows 10 machine that made it work. Which would imply that Microsoft is supporting macOS and Linux in some weird way? Who knows.

There isn't any way to support me, as there were quite a few people that worked to make this driver work. So there isn't exactly a great way to divvy out support. Not to mention that GitHub never really implemented any sort of good "charity" support. So I haven't ever really set anything up.

Really glad to hear that it's working for you. I guess I'll leave this open for a bit until I can get a second confirmation on firmware updating fixing the issue and then close this out with a note in the README.

the-t-in-rtf, 2019-04-03:
@FelixHenninger, that's an interesting idea, I'll have to try the firmware update once I'm home. For anyone else who doesn't have a working bootcamp partition or windows machine lying around, I tend to do windows-specific setup stuff in a [free virtual machine from MS](https://developer.microsoft.com/en-us/microsoft-edge/tools/vms/).  I do believe I did the firmware update the last time I commented, but that was more than six months ago, so fingers crossed.

the-t-in-rtf, 2019-04-04:
Very good news, after connecting my controller to a Windows VM and letting it go through its initial configure, I have the latest firmware, and the driver now works for me on High Sierra.

mortenboye, 2019-04-04:
I can confirm what @FelixHenninger and @the-t-in-rtf are reporting, that efter connecting and setting up on a Windows machine, it works on Mac (10.14.3) as a Xbox One Wired Controller. 
It can connect via Bluetooth, but does not appear in the Xbox360 control panel and doesn't seem to do anything.

the-t-in-rtf, 2019-04-04:
> It can connect via Bluetooth, but does not appear in the Xbox360 control panel and doesn't seem to do anything.

As far as I know that's totally normal, as the driver is only needed for wired connectors or connectors using the Xbox wireless USB adapter.  No driver is required when using bluetooth, at least not for the things I use my controller for.

FranticRain, 2019-04-04:
@mortenboye As @the-t-in-rtf said, the preference pane for this driver does not do anything for the Bluetooth connection of devices. The preference pane is designed to help you configure devices that are using the driver. Bluetooth connected devices are not using the driver in any way, and therefore do not show up in the preference pane.

This is all great news. I'll add some documentation on this in the README and then close up this issue. Glad this all got resolved.

closed 2019-05-13 by FranticRain