360Controller/360Controller #755: Post-install, non-responsive boot

velopike, 2018-09-18:
### Type of Controller
Replace this text with: Xbox One - Never connected

### OS Version
High Sierra

### Driver Version
v0.16.9

### Connection Method
Replace this text with: To be wired, never connected

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID:
<br/>Vendor ID:

### Response of Controller
Replace this text with information on how the physical controller reacts. Does the guide button light up? Does it vibrate? Etc.

### Response in Preference Pane
Replace this text with information about the controller's reaction in the preference pane. Does the controller appear in the preference pane? Do all of the buttons, joysticks, and triggers act as expected?

### Detailed Explaination of Issue
I have only just installed this driver and have not been able to restart/start my macbook pro since. The machine hangs on the loading screen with the user and loading bar.

FranticRain, 2018-09-18:
Please hold Command + V during boot and tell me what it hangs on.

velopike, 2018-09-19:
I remove it and reinstalled after doing a system restore, no issue on second install.

closed 2018-09-19 by FranticRain