360Controller/360Controller #757: PowerA Xbox One Controller Device Not Found

farrellse18, 2018-09-21:
I need help with this I saw that is said it was not compatible, but will it ever be. I have a PowerA controller i just bought it I did all the set ups step by step and i tried to do some of the coding that was in so feeds. Now this issue is mainly around the PowerA Xbox One controller. How can I set it up or are you working on patching it so PowerA is compatible?

farrellse18, 2018-09-21:
Product ID: 0x581a
Vendor ID: 0x24c6

FranticRain, 2018-09-22:
PDP and PowerA Xbox One controllers are incompatible with macOS 10.11+ thanks to a rewrite of the macOS USB kernel. We cannot resolve this issue. It is a bug in Apple's core OS code. There are currently no plans to proceed with a rewrite of the driver.

closed 2018-09-22 by FranticRain

tomlagier, 2020-06-30:
Just wanted to note that with the 1.0.0 line, the PowerA controller works.

NilsIrl, 2022-06-24:
Has this issue been reported to Apple? Where can I find the discussion/ticket (whatever apple's public bug tracker is)?