360Controller/360Controller #761: M

BUMPRW, 2018-09-29:
### Type of Controller
Xbox 360

### OS Version
10.13.6

### Driver Version
0.16.9, 0.16.8, 0.16.6, 0.12

### Connection Method
Wired

### Device Name and Info
identifies as MADCATZ 360 MC2:
Product ID:	0x4720
  Vendor ID:	0x0738  (Mad Catz, Inc.)
  Version:	22.31

### Response of Controller
Green lights on then off as described

### Response in Preference Pane
"No devices found" and Do you always have to enable the driver under advanced?

### Detailed Explaination of Issue
Appears the information for my Mad Cats 360 MC2 ring wheel doesn't match your Madcatz MC2.  Also I always have to check the box enable driver under the advanced tab and it comes up un- enabled after a restart.
I did attempt to edit info.plist.  It unlocked with permission and did save but

jrab66, 2018-10-01:
already added to the info.plist!

BUMPRW, 2018-10-01:
Roughly when will the change be available?  I’m not in a rush just curious.

BUMPRW, 2018-10-01:
Also, not sure if the vendor ID and Product ID that I provided match the change that I provided.  Is there another conversion table for these inputs?

18208<>4720 and 1848<>738

FranticRain, 2018-10-03:
Here's a build with this device added.

[Install360Controller.pkg.zip](https://github.com/360Controller/360Controller/files/2440526/Install360Controller.pkg.zip)

closed 2018-10-03 by FranticRain

BUMPRW, 2018-10-03:
Thanks, much appreciated. It appears to be working fine for both my controllers.  Just curious . . . can’t I just run the build.sh once you have committed or do I need to set-up my Xcode 10.0 for backward compatibility and go through the build steps?

FranticRain, 2018-10-03:
You have to set up your Xcode for backwards compatibility. The kext will not build against the current macOS APIs.