360Controller/360Controller #763: After install, freezes during restart, re-restarting fails

geoffhom, 2018-10-04:
### Type of Controller
Xbox One S Wireless, I think (Microsoft-booth giveaway, GDC 2018)

### OS Version
10.13.6

### Driver Version
16.9

### Connection Method
Wired/Bluetooth

### Device Name and Info
>Device Name: MacBook Air (11-inch, Mid 2012)

### Response of Controller
During install, the controller may have been connected via Bluetooth. (I was trying to get it to work while waiting for the install to finish.) The USB cable was also attached. 

### Response in Preference Pane
N/A

### Detailed Explaination of Issue
I have zero experience with recent Xbox controllers, but I got one for free at GDC 2018 from the Microsoft booth (https://news.xbox.com/en-us/2018/03/12/xbox-at-gdc-2018/). I decided to give the controller a try today. I tossed the box awhile ago, but I think it's an Xbox One S Wireless controller. It's blue. I could connect a USB cable from my Mac to the controller, then get it to connect to Mac's bluetooth (without installing any batteries), but the buttons seemed all wrong trying to set up Nuclear Throne on Steam. I had an old version of 360Controller (2013?) in my system, but that didn't see anything. So I downloaded v16.9. 

I tried to Install, then it said it couldn't install the extension without me granting access, but when I went to that pane, I couldn't see how to grant access. So I restarted because it said it was done anyway. After restart, nothing seemed different. So I tried installing 16.9 again. At some point, the Privacy pane showed that the extension was denied access, so I clicked to allow it. Then it looked like it had finished installing again, so I restarted again. 

But now the Apple logo would show up with the loading bar beneath, then it'd get real slow toward the end, then eventually get to 100%, but then stay stuck there. Soft resets didn't help. Hard reset didn't help. I read issue #751 and so went into repair mode (command-R during reset), then into Terminal, and tried all the uninstall commands you listed in #751. It didn't recognize the non-rm ones, and I had to find the right directories, but the rm ones worked. Then restart worked, at least this once (I'm writing this now in case I get locked out again). For good measure, after that restart I went into Terminal again and tried the rest of the uninstall commands:

    "Unload": "/Library/LaunchDaemons/com.mice.360Daemon.plist: No such file or directory
    "kextunload": "(kernel) Kext com.mice.driver.Xbox360Controller not found for unload request."
    "pkgutil --forget" worked

All that to say, there could be some serious bugs with this latest version. I don't know if it's because I installed over a 2013 version, or installed twice, or a permissions issue with when I clicked Allow for the extension, or just a bug with 16.9. I'm a little wary of trying it again. 

Also, in the release notes when it says, "Be sure to allow the dev in System Prefs," etc. I think that could be a lot clearer. Like more step-by-step. I went into Privacy & Security before installing and didn't see how to allow things. And during the first install, I still don't know what went wrong and why I didn't see the "allow" option. Did I need to wait longer? Was it because the pane was locked? 

Anyway, thanks for all your hard work.

FranticRain, 2018-10-04:
It's impossible to say for sure what caused your issues, but this hang on boot is consistent with issues found with the Bluetooth kext added to this driver around the time support for these Bluetooth style controllers was added. It's not impossible that you had a version that supported this device with those Bluetooth kexts still installed, and that's what caused the hang. There are a couple of things I can say for certain though:
- If you hadn't accepted the permissions, this would prevent the kext from loading, making it impossible to cause a boot hang
- Installing over a 2013 version shouldn't be a problem.
- Installing over the same version definitely isn't a problem.
- [There is a step by step guide on how to allow the device in the README, under the "Installation" section.](https://github.com/360Controller/360Controller#installation)

The only way I can find a solution to this problem and send you on your way with any useful solution, you'll need to try again, and if it does hang do this: Hard shut down your computer, and the next time, hold Cmd + V during launch. This will cause the computer to post logs, and eventually it will hang on something. That will let me know what went wrong. My best guess is that you won't experience the same issue again if you do re-install. But if there is an issue with this version of the driver, the only way to find out is if someone can do that for me.

geoffhom, 2018-10-05:
Thanks for the detailed and prompt reply. That encouraged me to try again. It seems to have installed fine. (No hang on restart.) It didn't ask for security access this time, I guess because I'd given it last time. I still need to see if the extension even recognizes my controller via USB; that part never worked (it said, "No Devices Found" even though the USB cable runs from my Mac to the controller, and I know that's how it gets power as the guide button lights up =), but I'll try that tonight when I get home. 

I hadn't heard of Verbose mode (command-V). Sorry I didn't use it when it was hanging!

Also, regarding a step-by-step guide for installation, that was helpful, but I didn't see it the first time (actually first two times) I installed. I was reading the Releases page (https://github.com/360Controller/360Controller/releases) and didn't know where else to look (and the text when running the installer didn't help). A simple fix would be to add a link to the step-by-step or README. That link could be added after "Make sure to allow access to the developer in System Preferences > Security & Privacy."

geoffhom, 2018-10-05:
Ok, I tried my controller and it works! I just plugged in the USB cable, the extension recognized it, all the buttons worked, and it worked in Steam with Nuclear Throne. Thank you! So, I'm still not sure what went wrong before. However, I recall that when I was following the advice for #751 and I was deleting files, I don't think they were all present. I think these were missing:

    sudo rm -rf /System/Library/Extensions/360Controller.kext
    sudo rm -rf /System/Library/Extensions/Wireless360Controller.kext
    sudo rm -rf /System/Library/Extensions/WirelessGamingReceiver.kext

Or at least the "Wireless" files. They were all present in `/Library/Extensions/`, I think.

I have nothing more to add. I'm not sure on issue etiquette, so feel free to close this when you see fit.

FranticRain, 2018-10-05:
It wouldn't particularly surprise me if the `/System/*` drivers were missing, as those are a vestige of some old stuff. Probably means that your driver version was much newer than 2013, though. Glad it all worked out for you.

closed 2018-10-05 by FranticRain