360Controller/360Controller #765: "No Device Found" In Preference Pane

irevoire, 2018-10-05:
### Type of Controller
Xbox 360 Pro Fight

### OS Version
MacOS Mojave 10.14

### Driver Version
Version 0.16.9

### Connection Method
Wired

### Device Name and Info
>Device Name: FightStick
<br/>Product ID: 0x187c
<br/>Vendor ID: 0x0079

### Response of Controller
Nothing happens.

### Response in Preference Pane
The preference pane show "No device found"

### Detailed Explaination of Issue
At first I though my issue was the same as this one : https://github.com/360Controller/360Controller/issues/736
But actually when I run `sudo kextload /Library/Extensions/360Controller.kext` nothing happens.

Also by searching for some log I've found that `dmesg` show something like that when connecting my controller :
```
FightStick: family specific matching fails
FightStick: family specific matching fails
FightStick: family specific matching fails
```
This line appear a **lot**.

But by scrolling I found there is also this thing ->
```
Report bytes: input:27 output:8 feature:8Collections : 2Buttons     : input:1 output:0 feature:0Values      : input:21 output:1 feature:1Element value capacity 1304Report count: 1Report ID: 0 input:216 output:64 feature:64HID: Legacy shim 2
001651.861825 IOUSBHostHIDDevice: IOUSBHostHIDDevice::interruptReadComplete: IO error 0xe00002ed, 9 retries remaining
001652.079732 IOUSBHostHIDDevice: IOUSBHostHIDDevice::interruptReadComplete: IO error 0xe00002ed, 8 retries remaining
```

I don't think it'll be usefull but the controller is being recognised via System Report > USB

Thanks in advance !

irevoire, 2018-10-05:
Also, in Hollow Knight when I connect the controller it pop up during half a second and then disappear

mainboy2006, 2018-10-09:
Try using a different cord to connect it, that worked for me

FranticRain, 2018-10-10:
This device has been added to the driver in 0.16.10. Since this looks like a Chinese knock-off controller, I'll give you the standard warning if it still doesn't work:

This controller is not supported by official Xbox hardware. As such, its support is outside the scope of this driver. We cannot and will not support this controller.

closed 2018-10-10 by FranticRain

kriku, 2020-02-04:
> Also, in Hollow Knight when I connect the controller it show up during half a second and then disappear

Have same issue with Teensy board in programming mode (have a loop when it show up and goes offline after retries)