360Controller/360Controller #768: Downloaded 16.9; Controller Still Will Not Connect

ButtonMapping, 2018-10-09:
### Type of Controller
Xbox One

### OS Version
10.13.6

### Driver Version
16.9

### Connection Method
Wired; UBS --> Micro USB (Supports data transferring)

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Xbox One Controller
<br/>Product ID: 0x02ea
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller
When plugged in, the controller vibrates, but does not light up in any way.

### Response in Preference Pane
No appearance in the preference plane.

### Detailed Explaination of Issue
I have read multiple threads on the subject of getting this driver to recognize the controller, but it still does not work.  Specifically, I checked to see that my controller appeared under the USB section of the System Report, and it does, but the driver still does not recognize the controller. Please help!

FranticRain, 2018-10-10:
Please run this command from your terminal and tell me what the output is:
```
sudo kextutil /Library/Extensions/360Controller.kext
```

ButtonMapping, 2018-10-10:
It says
Password: (Than a key emoji)
What am I doing wrong?

ButtonMapping, 2018-10-10:
Here you go:
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Kext rejected due to system policy: <OSKext 0x7feeac609920 [0x7fff8b43eaf0]> { URL = "file:///Library/StagedExtensions/Library/Extensions/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
Kext rejected due to system policy: <OSKext 0x7feeac609920 [0x7fff8b43eaf0]> { URL = "file:///Library/StagedExtensions/Library/Extensions/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
Diagnostics for /Library/Extensions/360Controller.kext:
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

FranticRain, 2018-10-10:
Please ensure that the driver has been allowed in System Preferences -> Security and Privacy -> General. This approval UI is only present in the Security & Privacy preferences pane for 30 minutes after you attempt to load the driver. Until you approve the kext, future load attempts will cause the approval UI to reappear but will not trigger another user alert.

[image]

ButtonMapping, 2018-10-10:
Thank you so much!

closed 2018-10-10 by ButtonMapping