360Controller/360Controller #774: Xbox One controller wired not appearing in the preference pane

SlightSarcasm, 2018-10-12:
### Type of Controller
Xbox One 

### OS Version
10.13.6

### Driver Version
16.10

### Connection Method
wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID: 0x02d1
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller
Light on

### Response in Preference Pane
Does not appear in 360 controller preference pane, but does appear in xone controller pane

### Detailed Explaination of Issue
My Xbox One controller isn't appearing in the preference pane, but it does appear in the system report. It also appears and responds to button presses in the xone driver, but this doesn't reflect on steam or in game.

FranticRain, 2018-10-12:
What do you mean "it does appear in the xone controller pane?" Are you saying that you have Xone-OSX installed?

SlightSarcasm, 2018-10-12:
> What do you mean "it does appear in the xone controller pane?" Are you saying that you have Xone-OSX installed?

I installed both drivers, when I was installing Xone-OSX I caught out of my eye that you had created a more up to date one.

FranticRain, 2018-10-13:
Please uninstall Xone-OSX. It should not be used, it's documented all over as such. If you install both at once, the operating system can only match to one driver at a time, and in this case, it's matched to Xone-OSX. Therefore it won't match to this driver. This is working as intended.

closed 2018-10-13 by FranticRain

SlightSarcasm, 2018-10-14:
Thanks much, basically TL;DR'd the install page oops..