360Controller/360Controller #776: Elite Controller Right Stick Deadzone Issue

itsfano, 2018-10-13:
### Type of Controller
Xbox Elite Controller (Brand New)

### OS Version
Sierra

### Driver Version
0.16.10

### Connection Method
Wired

### Response of Controller
Connects perfectly fine.

### Response in Preference Pane
Everything functions properly in preference pane.

### Detailed Explaination of Issue
While playing Fortnite, the reticle is unresponsive when making minor/precise adjustments near the center. These same slight motions register perfectly fine in the preference pane and even work within fortnite when I invert the sticks and switch aiming to the left stick. 

I've tried spoofing as a 360 controller as well as adjusting the slider in the preference pane to just about every point but continue to get a slight deadzone near the center.

FranticRain, 2018-10-13:
Do you have any means of testing this on an Xbox One console or Windows PC to verify that this is an issue with the driver? This sounds like a hardware issue to me.

itsfano, 2018-10-13:
Yes this is a brand new controller that works fine on an Xbox. Went out of town and decided to try it on gForce. Like I said, the motions are registered in the 360 Controller panel, just not on fortnite

FranticRain, 2018-10-13:
The driver does not provide any sort of compatibility for individual games beyond what is provided in the preference pane. The driver simply sends button data to the operating system. Once it does that, control is out of the driver's hands. Since Apple does not provide any standardized means of interacting with controllers in macOS, every game uses a different method to read controller data. We cannot provide individual fixes for each game within the controller. That simply isn't feasible. It is the responsibility of game developers to support controllers, not the other way around. We cannot and will not provide support for individual games. If you are using an Xbox One controller, we can recommend that you use the "Pretend to be an Xbox 360 Controller" option in the "Advanced" menu for better game compatibility. Beyond that, there is nothing we can do.

closed 2018-10-13 by FranticRain