360Controller/360Controller #780: Controller Not Recognized in Preferences

j-h-m, 2018-10-21:
### Type of Controller
Xbox 360 Wired - Gamestop Model

### OS Version
10.13.6

### Driver Version
0.16.10

### Connection Method
Wired

### Device Name and Info
>Device Name: Gamepad for Xbox 360:
<br/>Product ID: 0x0401
<br/>Vendor ID: 0x0e6f
<br/> etc.
  Version:	1.00
  Serial Number:	069FDC43
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Performance Designed Products
  Location ID:	0x14100000 / 7
  Current Available (mA):	500
  Extra Operating Current (mA):	0

### Response of Controller
Ring around main button lights up for approx 1 second and then goes off. No response in any programs.

### Response in Preference Pane
Does not appear, no response.

### Detailed Explaination of Issue
No response from my controller after installing the driver version 0.16.10, latest release. I am assuming this is because it is a 3rd party controller (gamestop 360 wired gamepad).

P.S --- I really like the way you guys have a template for issues, this is awesome! 👍

FranticRain, 2018-10-21:
Your controller is already added to the driver as `GameStopGamepad3`, so it should be recognized. 

Please ensure that the driver has been allowed in System Preferences -> Security and Privacy -> General. This approval UI is only present in the Security & Privacy preferences pane for 30 minutes after the alert. Until you approve the kext, future load attempts will cause the approval UI to reappear but will not trigger another user alert.

[image]

If you've already done this, then go into your terminal and load the kext manually to see if you get an error message.
```
sudo kextutil /Library/Extensions/360Controller.kext
```

j-h-m, 2018-10-27:
I reinstalled the latest release and allowed the application in security preferences and that worked. It is now showing up, thanks @FranticRain! Marking this as closed.

Most applications ask / notify you of this on first load, I must have missed that the first time... but I don't understand why it wouldn't ask again, I just recently started using a Mac though, so I don't have much experience.

[image]

closed 2018-10-27 by j-h-m