360Controller/360Controller #782: Driver will not stay enabled after reboot

SuperSirLink, 2018-10-22:
### Type of Controller
Official Microsoft Xbox 360 Wireless

### OS Version
10.14

### Driver Version
0.16.10
n/a

### Response of Controller
lights flash

### Response in Preference Pane
no controllers found

### Detailed Explanation of Issue
After doing a clean install of Mojave and reinstalling the driver, I found the enable driver under the advanced tab was not checked...  So I checked enable and rebooted.  But the setting was back to being unchecked once I logged back in.  Before I enable it, the adapter doesn't show up in system profiler... Once I enable it, it does but the controller doesn't connect.  I am assuming it needs to be enabled on startup, but every startup returns it to not enabled status.

Querying the kext shows:
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

FranticRain, 2018-10-22:
Your device information is not N/A. The "enabled" checkbox is a symptom of the issue, not an indicator. Add your device's information before I can help you.

SuperSirLink, 2018-10-23:
Sorry, I miss understood what it was referring to... I thought you only needed that for non official controllers...

FranticRain, 2018-10-24:
As of macOS 10.11, Wireless controllers have been unsupported by this driver. This is thanks to a rewrite of the macOS USB kernel. This issue can only be resolved through a rewrite of the driver, which is currently unplanned. Therefore I will be closing this issue, since it is outside the realm of supported devices

What you did here is load the wired controller driver, which was working fine all along. There is no way to turn a wireless 360 controller into a wired controller via a cable adapter such as a "Play and Charge" kit. Therefore, this device is unsupported by this driver.

closed 2018-10-24 by FranticRain

SuperSirLink, 2018-10-24:
That’s odd, because it was working fine under 10.13