360Controller/360Controller #783: Will not recognize controller

bradels12, 2018-10-24:
### Type of Controller
XBOX One - Amazon Basics Model 1500527-01

### OS Version
MacOS High Sierra 10.13.6

### Driver Version
0.16.10

### Connection Method
Wired

### Device Name and Info
Device Name: Xbox ONE Pro Ex controller
Product ID: 0x543a
Vendor ID: 0x24c6

### Response of Controller
Controller is not recognised

### Response in Preference Pane
No response, is not detected

### Detailed Explanation of Issue
This is a 3rd party controller, wired USB and is not recognized. I read the readme article, but I am not technical enough to make a new build of the kext file. Would someone be able to add my controller type and build a kext file for me. At that point, I am confident that I will be able to run the scripts to get it loaded into the OS.

Thanks

FranticRain, 2018-10-25:
PDP and PowerA Xbox One controllers are incompatible with macOS 10.11+ thanks to a rewrite of the macOS USB kernel. We cannot resolve this issue. It is a bug in Apple's core OS code. This controller is created by PDP and re-branded as an Amazon controller.

closed 2018-10-25 by FranticRain

bradels12, 2018-10-25:
Bummer, I purchased this controller specifically to use with my Mac (did
not realise it was  a PDP controller) I will return it. Is there a
controller you would recommend?

Thanks

FranticRain, 2018-10-25:
I personally only buy first party controllers. I don't know what's good in 3rd party controllers.

bradels12, 2018-10-25:
Will this one work? It is an original Microsoft 4N6-00001 Xbox Controller +
Cable Windows
<https://www.amazon.com/Microsoft-4N6-00001-Controller-Cable-Windows/dp/B01LPNKGGI/ref=sr_1_2?s=videogames&ie=UTF8&qid=1540437064&sr=1-2&keywords=xbox+one+wired+controller&refinements=p_89%3AMicrosoft>
?

FranticRain, 2018-10-25:
That controller should work, but given the current issues with Amazon selling knock off versions of that controller under that name, I can't promise anything.