360Controller/360Controller #784: Controller does not work reliably.

matthewmcvickar, 2018-10-25:
### Type of Controller
Xbox 360, Xbox One

### OS Version
macOS 10.13.6

### Driver Version
v0.16.10

### Connection Method
Wired

### Device Name and Info
Device Name: Microsoft Xbox One Wired Controller
Product ID: 0x02ea
Vendor ID: 0x045e

### Response of Controller
After plugging in, the controller rumbles and then the light starts blinking.

### Response in Preference Pane
Controller does not appear in the pref pane. It just says 'No devices found'

### Detailed Explanation of Issue
The pref pane does not recognize the controller.

This is the output of  `sudo kextutil /Library/Extensions/360Controller.kext`:

```sh
Untrusted kexts are not allowed
Kext with invalid signature (-67062) denied: /Library/StagedExtensions/System/Library/Extensions/60BF9155-0B0E-49AE-BDFC-E13C856A1086.kext
Bundle (/System/Library/Extensions/RemoteControl.USBLAN_usbpart.kext) failed to validate, deleting: /Library/StagedExtensions/System/Library/Extensions/60BF9155-0B0E-49AE-BDFC-E13C856A1086.kext
Unable to stage kext (/System/Library/Extensions/RemoteControl.USBLAN_usbpart.kext) to secure location.
Untrusted kexts are not allowed
Kext with invalid signature (-67062) denied: /Library/StagedExtensions/System/Library/Extensions/F3A9DFB8-3E9A-47B5-8C25-612B2DD4DD0C.kext
Bundle (/System/Library/Extensions/Belcarra.USBLAN_usbpart.kext) failed to validate, deleting: /Library/StagedExtensions/System/Library/Extensions/F3A9DFB8-3E9A-47B5-8C25-612B2DD4DD0C.kext
Unable to stage kext (/System/Library/Extensions/Belcarra.USBLAN_usbpart.kext) to secure location.
Untrusted kexts are not allowed
Kext with invalid signature (-67062) denied: /Library/StagedExtensions/System/Library/Extensions/71B53FAD-F355-4AD2-8001-EDC31B30B931.kext
Bundle (/System/Library/Extensions/Belcarra.USBLAN_netpart.kext) failed to validate, deleting: /Library/StagedExtensions/System/Library/Extensions/71B53FAD-F355-4AD2-8001-EDC31B30B931.kext
Unable to stage kext (/System/Library/Extensions/Belcarra.USBLAN_netpart.kext) to secure location.
Untrusted kexts are not allowed
Kext with invalid signature (-67062) denied: /Library/StagedExtensions/System/Library/Extensions/BF887CFF-F36F-4A7E-9F22-B74C9718E3B5.kext
Bundle (/System/Library/Extensions/NovationUSBAudio.kext) failed to validate, deleting: /Library/StagedExtensions/System/Library/Extensions/BF887CFF-F36F-4A7E-9F22-B74C9718E3B5.kext
Unable to stage kext (/System/Library/Extensions/NovationUSBAudio.kext) to secure location.
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings:
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done):
        ChatPadKeyboardEvents
```

FranticRain, 2018-10-25:
Is this issue only with 0.16.10, or have you experienced it with older versions of the driver as well?

exic5, 2018-10-28:
are you sure that when you downloaded it you opened the Security & Privacy pane and allowed the developer on your device???

matthewmcvickar, 2018-10-29:
I am certain that allowed the “software from Developer ‘Drew Mills’’ to be installed.

I haven't had time to try an older version yet, but I will try that later.

Until then, I wanted to let you know a new wrinkle in this. It doesn't work at all when I restart my computer, but once I run `sudo kextutil /Library/Extensions/360Controller.kext` in the Terminal and plug the controller back in, it *is* recognized.

FranticRain, 2018-10-30:
Since you were able to use the driver correctly, even just once, it is obviously not a signing certificate issue. The interesting part is this new information, though. Basically what you've told me is that the driver works, and the operating system is failing to load it properly. Since we don't touch the `probe` function, it means that this issue is 100% outside the control of the driver. Things I would consider possible causes:
- A bad cable
- An install that had gone bad
- Some other issue with your operating system

I'm going to go ahead and put a `wontfix` tag on this for now, since it seems to be on your end, but I'll keep it open in case you need further assistance.

matthewmcvickar, 2018-11-07:
Now that I have installed OS X Mojave, this problem is no longer happening. The controller is recognized every time regardless of restart. I guess it was a problem with the kexts. ¯\_(ツ)_/¯

closed 2018-11-07 by matthewmcvickar