360Controller/360Controller #786: Logitech F310 it's not found

tracy-ridwan, 2018-10-28:
### Type of Controller
Logitech F310

### OS Version
10.14 MacOS Mojave

### Driver Version
Version 0.16.10

### Connection Method
wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID: 0xc216
<br/>Vendor ID: 0x046d  

### Response of Controller
The controller is conected but can't be discover by the driver

### Response in Preference Pane
Replace this text with information about the controller's reaction in the preference pane. Does the controller appear in the preference pane? Do all of the buttons, joysticks, and triggers act as expected?

### Detailed Explaination of Issue
Replace this text with a detailed explaination of your issue. Simply saying "it doesn't work" is a quick way to ensure that your issue is never fixed. Without detailed information on your specific issue, we cannot fix your issue. Including as much information as possible in this section increases the likelyhood of getting your issue quickly and effectively resolved.

FranticRain, 2018-10-30:
Here is a version of the driver built with your device added. Sorry for the .zip, Github doesn't let you upload .pkg files. Just make sure when using an F310 that if you want to use it in XInput mode. This means flipping the switch on the back of the controller over to "X" from "D".

[Install360Controller.pkg.zip](https://github.com/360Controller/360Controller/files/2527750/Install360Controller.pkg.zip)

tracy-ridwan, 2018-10-30:
@FranticRain 
Hi, i already tried it and it's not working
is there anything that i installed wrongly?
-i already check that my controller is connected via system report
-i installed the zip that you attach above
-i double check the product id and vendor id is match also (i tried to convert it to decimal from the hex it's match too)
-the certificate for installation already being provided also

is there anything that i'm missing?

FranticRain, 2018-10-31:
Please run this command in your terminal, enter your password when prompted, and tell me what the output is:
```
sudo kextutil /Library/Extensions/360Controller.kext
```

tracy-ridwan, 2018-10-31:
`/Library/StagedExtensions/Library/Extensions/0E266DEE-370A-41B0-862A-3A2B55DE1178.kext does not appear in strict exception list for architecture: x86_64
Untrusted kexts are not allowed
Kext with invalid signature (-67007) denied: /Library/StagedExtensions/Library/Extensions/0E266DEE-370A-41B0-862A-3A2B55DE1178.kext
Bundle (/Library/Extensions/BJUSBLoad.kext) failed to validate, deleting: /Library/StagedExtensions/Library/Extensions/0E266DEE-370A-41B0-862A-3A2B55DE1178.kext
Unable to stage kext (/Library/Extensions/BJUSBLoad.kext) to secure location.
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents`

FranticRain, 2018-11-01:
I just pulled out my controller and tested it against the zipped version of the driver that I gave you, and it works just fine. So the problem is on your side. Here's all of the advice I can give you:
1. Reinstall the driver from the zip I gave you. It's possible that you installed some weird unsigned version of the driver, and that definitely won't work.
2. Make sure you did the following:

Please ensure that the driver has been allowed in System Preferences -> Security and Privacy -> General. This approval UI is only present in the Security & Privacy preferences pane for 30 minutes after the alert. Until you approve the kext, future load attempts will cause the approval UI to reappear but will not trigger another user alert.

[image]

tracy-ridwan, 2018-11-01:
[image]
[image]

Hi could you check my screen shoot...
i already tried to reinstall again using your zip
but it's still doesn't work..

[image]

i can't find it...

but the things that make me worried that the certificate pop up that you put above it's not appearing but when i check the certificate it's already inside my keychain..
is it okay ?

FranticRain, 2018-11-02:
So looking at that log at closer, you'll notice that `360Controller.kext` actually succeeds in loading. What fails are other kexts on your system. I don't know if you're running a Hackintosh or something else, but your system is the one failing. This has nothing to do with the driver, because it loads just fine. The only advice I can offer you is to make sure that your controller is in XInput mode, not DirectInput mode. The other kext loading issues are not related to this driver, but will prevent it from working properly.

closed 2018-11-02 by FranticRain

yuanzimingozm, 2022-09-18:
sorry,don`t kown ,this error is what

Error Domain=KMErrorDomain Code=71 "Incompatible architecture: Binary is for x86_64, but needed arch arm64e
Incompatible architecture: Binary is for x86_64, but needed arch arm64e
Unsupported Error: one or more extensions are unsupported to load 	Kext com.mice.driver.Xbox360Controller v0.16.11 in executable kext bundle com.mice.driver.Xbox360Controller at /Library/Extensions/360Controller.kext
Unsupported Error: one or more extensions are unsupported to load 	Kext com.mice.driver.Xbox360Controller v0.16.11 in executable kext bundle com.mice.driver.Xbox360Controller at /Library/Extensions/360Controller.kext" UserInfo={NSLocalizedDescription=Incompatible architecture: Binary is for x86_64, but needed arch arm64e
Incompatible architecture: Binary is for x86_64, but needed arch arm64e
Unsupported Error: one or more extensions are unsupported to load 	Kext com.mice.driver.Xbox360Controller v0.16.11 in executable kext bundle com.mice.driver.Xbox360Controller at /Library/Extensions/360Controller.kext
Unsupported Error: one or more extensions are unsupported to load 	Kext com.mice.driver.Xbox360Controller v0.16.11 in executable kext bundle com.mice.driver.Xbox360Controller at /Library/Extensions/360Controller.kext}