360Controller/360Controller #788: Xbox One Controller Not Recognized in Preference Pane

BiffCannon, 2018-10-31:
### Type of Controller
Xbox One

### OS Version
Mojave 10.14

### Driver Version
0.16.10

### Connection Method
 Wired

### Device Name and Info
>Device Name: Bluetooth USB Host Controller
<br/>Product ID: 0x8294
<br/>Vendor ID: 0x05ac (Apple Inc.)

### Response of Controller
Xbox button blinks

### Response in Preference Pane
Device Not found

### Detailed Explaination of Issue
Installed driver, restarted, xbox controller appears in preference pane. I click into those preferences and turn on the wired controller, but it just blinks and the device is never "found". I've reinstalled it a couple different times, but it never recognizes the controller. I've checked "EnableDriver"  In 'Advanced', but that doesn't help. Assistance in troubleshooting will be greatly appreciated. Thank you.

FranticRain, 2018-11-01:
So you see how the vendor id of your "Bluetooth USB Host Controller" is labeled as "Apple Inc?" That's pretty clearly not your controller. It's an internal part of your computer. Try and find your controller again and let me know when you get the right vendor and product IDs, so I can assist you with this issue.

BiffCannon, 2018-11-02:
It's not listed under hardware>usb then.

FranticRain, 2018-11-02:
As noted in the README, you are likely using what is referred to as a "charge" cable with your controller. These types of cables don't have the data pins connected and therefore, cannot be recognized by your system. Please try a different cable and see if you can get your controller to show up and get your vendor and product IDs. The driver will not work unless your controller is visible in this menu. https://github.com/360Controller/360Controller#xbox-one-controllers-connected-with-usb

FranticRain, 2019-01-15:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-01-15 by FranticRain